cruise-automation/webviz #374: [feature request] support multiple remote-bag-urls.

jasonimercer, 2020-03-19:
Would it be possible to support multiple remote bags (even 2) in the same way that you support 2 local bags?

Our bagging strategy breaks data into different bags, a "main" bag with most general information and then more customized bags that are useful for triaging specific issues.

Our triage teams have become used to using the webviz links our backend data collector provides, these links bring them to our instance of webviz with a remote-bag-url pointing at a share of the converted data. I would like to be able to offer them links that will present them with simultaneous views into the main bag + the planner image diagnostic bag + a layout url so they're all setup to diagnose whatever they see.

I had a read through the code and I feel like your architecture is very close to already supporting this. There may be issues with multiple data caches? I'm really not sure. If this were a C++ project, I'd be happy to submit an MR but my js skills are classified as "detrimental to the planet".

jtbandes, 2020-03-19:
Yes, in fact we are also tracking this request internally. (It's become more relevant in today's work-from-home climate.) I don't want to promise a timeline on behalf of the folks actually working on it, but I think it's likely to be implemented soon, so stay tuned! Thanks for the request!

cjds, 2020-03-26:
+1 would also love the ability for multiple remote bag URLs

jtbandes, 2020-03-26:
It's already covered in #378!

janpaul123, 2020-04-09:
This is in the latest release! #396 Thanks for the request!

closed 2020-04-09 by janpaul123