cruise-automation/webviz #376: [feature request] change panel background color

agarret7, 2020-03-23:
Would it be possible to add functionality for changing the background color of the 3D visualization panels? In particular it would be nice to be able to select a white background.

janpaul123, 2020-03-23:
Sounds like a good idea to me! cc @vidaaudrey @davidswinegar

vidaaudrey, 2020-03-23:
Yeah good idea. We could add as part of the 3D panel menu options [here](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/TopicGroups/LayoutForTopicGroups.js#L462-L476), perhaps with a color picker to choose the background. @agarret7  let us know if you are interested in taking it? Otherwise I could pick it up early next week.

agarret7, 2020-03-27:
I'm receiving a bunch of compile  when running `npm run bootstrap`, mostly looking something like:

    In file included from ../../nan/nan_new.h:189:0,
                     from ../../nan/nan.h:222,
                     from ../src/binding.cpp:1:
    ../../nan/nan_implementation_12_inl.h: In static member function ‘static Nan::imp::FactoryBase<v8::Function>::return_t Nan::imp::Factory<v8::Function>::New(Nan::FunctionCallback, v8::Local<v8::Value>)’:
    ../../nan/nan_implementation_12_inl.h:105:32: error: no matching function for call to ‘v8::Function::New(v8::Isolate*&, void (&)(const v8::FunctionCallbackInfo<v8::Value>&), v8::Local<v8::Object>&)’
                               , obj));

agarret7, 2020-03-27:
I accidentally closed the issue.

vidaaudrey, 2020-03-27:
Could you maybe try node v10.15.3 (the version I have at my Mac)?

agarret7, 2020-03-27:
That worked. Perhaps worth opening another issue on compatibility with `nodejs: '>10.19.0'`?

vidaaudrey, 2020-03-27:
Yeah totally.