cruise-automation/webviz #381: Not able to get S3 bag data to show up in Webviz using presigned URLs

cjds, 2020-03-26:
Hello,

I tried using a presigned URL from S3 to put my bag into webviz, but all I got was the following

```
Error: Remote file does not support HTTP Range Requests. Often this is due to missing CORS headers on the requested URL: https://*******/navigate/2020-01-31-13-18-31.376118_task-34984_action-204797_freight100-1173_0.bag?AWSAccessKeyId=****&Signature=*******&Expires=1585244993
```

This is the stacktrace the issue is coming from.
```
    at exports.default.open (https://webviz.io/dist/WorkerDataProvider.worker.js?e33bad3f1753231ae64e:621:1081)
    at async exports.default.open (https://webviz.io/dist/WorkerDataProvider.worker.js?e33bad3f1753231ae64e:711:1838)
    at async exports.default.initialize (https://webviz.io/dist/WorkerDataProvider.worker.js?e33bad3f1753231ae64e:543:2728)
    at async exports.default.initialize (https://webviz.io/dist/WorkerDataProvider.worker.js?e33bad3f1753231ae64e:452:2034)
```

In the network tab, everything looks like a 200.

janpaul123, 2020-03-26:
Ah yeah, usually you just need to set the right CORS settings on the S3 bucket. We have an example Terraform configuration [here](https://github.com/cruise-automation/webviz/blob/7cc88e02e0e8270c98ce56978344f2950d8d803c/packages/webviz-core/src/util/corsError.js#L18-L23).

cjds, 2020-03-26:
Got it in one. It was indeed a CORS issue.

For future reference for people using S3 this is the CORS configuration I used (it might be a little more loose than the minimum you require to get webviz working)

```
<?xml version="1.0" encoding="UTF-8"?>
<CORSConfiguration xmlns="http://s3.amazonaws.com/doc/2006-03-01/">
<CORSRule>
    <AllowedOrigin>*</AllowedOrigin>
    <AllowedMethod>HEAD</AllowedMethod>
    <AllowedMethod>GET</AllowedMethod>
    <AllowedMethod>PUT</AllowedMethod>
    <AllowedMethod>POST</AllowedMethod>
    <AllowedMethod>DELETE</AllowedMethod>
    <AllowedHeader>*</AllowedHeader>
    <ExposeHeader>ETag</ExposeHeader>
    <ExposeHeader>Content-Type</ExposeHeader>
    <ExposeHeader>Accept-Ranges</ExposeHeader>
    <ExposeHeader>Content-Length</ExposeHeader>
</CORSRule>
</CORSConfiguration>
```

closed 2020-03-26 by cjds