cruise-automation/webviz #390: [feature request]Visualization Pose Array/Objects List in 3D Panel

Shiming94, 2020-04-01:
We intend to visualize the dynamical positions of detected objects from camera or radar. Usually the number of the member in the object array is not fixed, and the array is expressed in message type PoseArray. Is that possible to support the visualization of message type PoseArray in 3D Panel?

janpaul123, 2020-04-01:
It looks like we currently don't support that, but it might be easy to add around here https://github.com/cruise-automation/webviz/blob/a3fc95267f9eb68530cbb0da27a1eb912032ee71/packages/webviz-core/src/panels/ThreeDimensionalViz/SceneBuilder/index.js#L457 Feel free to take a shot at it if you want, we can help if you run into issues.

Shiming94, 2020-04-02:
Thank you very much. I will try it.

Shiming94, 2020-04-03:
> It looks like we currently don't support that, but it might be easy to add around here [...]

Hi, I have a further question. Can I also print the lane information in message type geomety_msgs/Polygon in 3D Panel, or I still need to add something?

Thank you very much.

Shiming94, 2020-04-06:
Yes. I just want to print like the right/left lane boundaries and center line by Polygon.

Shiming94, 2020-04-06:
Oh, Sorry, I think I find the solution. The message type Polygon can not be printed, but PolygonStamped can.  Thanks a lot.

closed 2020-10-19 by vidaaudrey