cruise-automation/webviz #391: Changing to predefined layouts without reloading webviz

ubicray, 2020-04-01:
So I'm trying to modify webviz to make it more touch interface friendly
One part is to have some predefined layouts that can be switched by tapping some buttons
So far what has worked for me is to use the importLayout in PlayerManager and basically add some other layouts than the demo one to load those layouts. But as it's taking URL parameters, it takes reloading to take into effect so that's not great

How would I get it to work so that I could load some predefined panel layouts without having to reload webviz? Anybody who's done something similar?

janpaul123, 2020-04-01:
You should be able to call the `importPanelLayout` Redux action (which in `PlayerManager` appears to get renamed to `importLayout`) at any time to change the layout, without a reload. So you could call that action when clicking a button.

If you want to contribute this functionality back to Webviz, maybe it would be cool if one can add a `?preset-layouts=https://example.org/my-presets.json`, and then in that JSON file maybe you can specify all the layouts, e.g. `{ "layouts": [{ "name": "Some name", layout: { … } }, …] }`. What do you think?

vidaaudrey, 2020-10-19:
@ubicray  pls refer [importPanelLayout](https://github.robot.car/cruise/webviz/blob/master/webviz-core/src/reducers/panels.js#L392-L411) for implementation. You could render different layouts in the menu, and call `importPanelLayout` after the user clicks the menu items.  Gonna close the issue now and let me know if you have any questions.

closed 2020-10-19 by vidaaudrey