cruise-automation/webviz #392: Transform lookup / Camera projection in Node Playground

jackma, 2020-04-06:
Hello, Webviz!

As far as I can tell, it looks like Node Playground doesn’t support TF lookups or camera projections out of the box. Do you have any advice on dealing with TFs and projections in Node Playground?

I also noticed Webviz has [support for transforms](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/Transforms.js). But it doesn't look like it can be imported.

janpaul123, 2020-04-06:
Oh hey Jack! 😄 

Yeah, we currently don't offer any library support in Node Playground so you'd have to copy/paste some of those functions into your nodes yourself. We do have plans on adding library functions though. cc @troygibb @davidcrawford we should consider including TF lookups and camera projections when we add Node Playground libraries.

jackma, 2020-04-06:
Ah. Thanks for the consideration and ultrafast response!

janpaul123, 2020-06-09:
Btw I set up a Slack workspace to make it easier for us to chat; join us using the link here: #461