cruise-automation/webviz #398: remote-bag will not play if remote-bag-url-2 points at an empty bag

jasonimercer, 2020-04-13:
In our workflow we are trying to have the 2nd bag be planner diagnostic images but if the robot has not made a plan during the event that was recorded, Webviz will not let us look at the group of primary bag + empty secondary bag.

janpaul123, 2020-04-13:
Weird, do you get some error message or so? In what way does it not work? Do you have example bags that we could reproduce with?

jasonimercer, 2020-04-13:
Yup, there is an error message (thanks for surfacing it so clearly):

```
Error initializing player
Error: CachedFilelike#read invalid input
    at CachedFilelike.read (webpack-internal:///./packages/webviz-core/src/util/CachedFilelike.js:114:13)
    at BagReader.readConnectionsAndChunkInfo (webpack-internal:///./packages/webviz-core/node_modules/rosbag/dist/web/index.js:3287:16)
    at eval (webpack-internal:///./packages/webviz-core/node_modules/rosbag/dist/web/index.js:3321:12)
    at new Promise (<anonymous>)
    at BagReader.readConnectionsAndChunkInfoAsync (webpack-internal:///./packages/webviz-core/node_modules/rosbag/dist/web/index.js:3320:12)
    at Bag.open (webpack-internal:///./packages/webviz-core/node_modules/rosbag/dist/web/index.js:3966:38)
    at async BagDataProvider.initialize (webpack-internal:///./packages/webviz-core/src/dataProviders/BagDataProvider.js:99:7)
    at async ApiCheckerDataProvider.initialize (webpack-internal:///./packages/webviz-core/src/dataProviders/ApiCheckerDataProvider.js:75:34)
```
[robot_ph_cpe22_60_FCA66B26_planner_image_2_webviz.bag.gz](https://github.com/cruise-automation/webviz/files/4471387/robot_ph_cpe22_60_FCA66B26_planner_image_2_webviz.bag.gz)