cruise-automation/webviz #400: LCM support

lhcubillos, 2020-04-15:
Hi! Are there any plans to extend WebViz to support LCM? (https://lcm-proj.github.io/). If not, do you have any advice on how to adapt the app to LCM? It works in a very similar way as ROS, and has the equivalent of ROSbags as well.

janpaul123, 2020-04-15:
Currently not, and internally we're pretty tied to the binary format of ROS bags. Are there any conversion tools from their data format to ROS bags? That might be the easiest approach.

lhcubillos, 2020-04-16:
They don't have a native one, but it's probably not that big a challenge, so thank you for the advice! What about the live feed (not from rosbags)? Do you think it would be possible to adapt it to LCM? LCM works using UDP multicast, with no master node.

janpaul123, 2020-04-16:
That might be a bit harder, but once you have some code to convert from LCM to the ROS message format it shouldn't be too hard. You could write a ROS node that subscribes to the LCM messages, converts them to ROS messages, and publishes them to a ROS master (ideally on different topics). Then you'd only have to boot that ROS node, a ROS master, and the rosbridge, and it should hopefully all work?

So yeah, the biggest step would probably be doing that initial conversion to the bag format.

lhcubillos, 2020-04-16:
Thank you! That is a very good idea.

closed 2020-04-16 by lhcubillos

janpaul123, 2020-04-16:
Cool, keep us posted on how it goes! Would be cool if we can link future people with the same problem to whatever you come up with. 😃