cruise-automation/webviz #402: Plot panel not working on demo

lhcubillos, 2020-04-17:
Hi! I'm trying to use Webviz to visualize information from an underwater ROV, but I'm having problems with the Plot panel: it is not actually plotting the data. I tried the demo, at https://webviz.io/app/?demo, and it doesn't work as well. I'm in macOS 10.15.3, and the Chrome version I'm using is `Version 80.0.3987.163 (Official Build) (64-bit)`. I attach a screenshot of the problem, using Incognito Mode to try to isolate it. 
[image]

lhcubillos, 2020-04-17:
I updated Chrome to `Version 81.0.4044.113` but the problem remains.

janpaul123, 2020-04-17:
Thanks for letting us know, we totally missed this! https://github.com/cruise-automation/webviz/pull/399 should fix this, but is currently failing CI — @hassoncs is going to look into this today.

hassoncs, 2020-04-20:
@lhcubillos, this should be fixed now. Sorry that happen and thank you for reporting it!

closed 2020-04-20 by hassoncs

lhcubillos, 2020-04-20:
I don't know if I am the problem now, but I still can't see the plotting happening in the demo. 
[image]

I tried erasing the cache and cookies, and working in Incognito, but to no success.

janpaul123, 2020-04-20:
It took some time to build, but it looks good for me now, can you check again?

lhcubillos, 2020-04-20:
Now it's working! Thank you very much!

janpaul123, 2020-05-05:
@lhcubillos I've been reaching out to Webviz users to get to know the community better, and I'd love to learn more about the underwater robots that you use Webviz for. Would you be open to do a quick video call with me some time? If so, can you email me at jp.posma@getcruise.com? 😃