cruise-automation/webviz #406: Run webviz from source build without launching docs page

ben-greenberg, 2020-04-21:
I've built webviz from source and have been using it with great success. The only kink in my system is that currently when I want to open up the webviz application in a browser, I have to run ```npm run docs``` which launches the docs app in addition to running the webviz application.

Is there another way to run just the application without launching the docs page?

janpaul123, 2020-04-22:
You could host it online somewhere? We host webviz.io/app using Github Pages.

castaway2000, 2020-04-25:
@janpaul123 I think what they are saying is that they would like to run the webviz App as landing page. Is there an easy way to implement the webviz-core package in a standard html page or modify the existing React application to point the landing page to be the webviz app? 

I was also looking into doing this using the webviz-core package as a static website however it seems the webviz-core package is not able to source /dist/webvizCoreBundle.js and is not in the repo from what I can tell, How is this bundle sourced?

if the bundle was included it would make for an easier solution to the problem by simply running the package as a standalone app.

janpaul123, 2020-04-28:
You can see here how we publish our [`gh-pages`](https://github.com/cruise-automation/webviz/tree/gh-pages) branch: https://github.com/cruise-automation/webviz/blob/6623f080b2bee11144df1bb70c2ac93db51a37fa/package.json#L135

You can also statically host the gh-pages branch directly yourself. It's what gets hosted on webviz.io. You can see that it has an `app/index.html` that contains the actual application.

janpaul123, 2020-04-28:
Maybe we should pull that out into its own command so everyone can easily generate a static build? Happy to take PRs for that!

kevinrev26, 2020-05-05:
What if I wanted to use the individual components of the WebViz Core?

How should I import them?

janpaul123, 2020-05-05:
We don't currently support that. Those are considered internal API and might change heavily on every release. But if you really want to I guess you can just `import` them like any other component; e.g. `import useDataSourceInfo from "webviz-core/src/PanelAPI/useDataSourceInfo";`

janpaul123, 2020-06-09:
Btw I set up a Slack workspace to make it easier for us to chat; join us using the link here: #461