cruise-automation/webviz #410: How to load a local bag file without dragging and dropping the file?

castaway2000, 2020-04-27:
Hello I am spinning up a server of my own that is connected to a GUI full of referenced bag files, I am using the service to click on the bag file and load it into webviz without needing to have the bag file on my laptop. How would I easily load a bag-file into webviz without having a rosbridge to stream it or a bagfile to drop into it? For example your demo uses a file stored in an s3 bucket, how would someone like me achieve the same kind of result?

janpaul123, 2020-04-28:
It should work with any server that supports range requests and CORS. We have some [instructions](https://github.com/cruise-automation/webviz/blob/6623f080b2bee11144df1bb70c2ac93db51a37fa/packages/webviz-core/src/util/corsError.js#L9-L36) for setting up S3 and GCS buckets, which might be helpful in learning how to configure your server. What particular server software do you use?

There's a thread here #247 with some people who attempted something similar, but unfortunately they never posted back their configs. I'll ask them again there.

castaway2000, 2020-04-28:
@janpaul123 this is really cool info, thank you for sharing! Is there a a same origin process for loading local bag files?

janpaul123, 2020-04-28:
No for local bag files we get a [File handle](https://developer.mozilla.org/en-US/docs/Web/API/File), which gives us direct random access into the file. The `?remote-bag-url=…` option is substantially more complicated since it works over HTTP and we have to do all sorts of buffering ourselves.

janpaul123, 2020-04-28:
I would recommend hosting your bag files in S3 or GCS though, instead of on your own server, because it's just so much easier to get working. Another option is to use sshfs or so to connect to your remote file system and drag in the files to Webviz that way. But yeah, if you really want to stream the files directly from your server, hopefully the stuff I linked above is helpful.

janpaul123, 2020-04-29:
Cool, so looks like if you use Flask you can use this: https://github.com/cruise-automation/webviz/issues/247#issuecomment-621175154

closed 2020-06-14 by castaway2000