cruise-automation/webviz #412: Loader splash image persists after remote bags load / webviz initializes.

jasonimercer, 2020-04-29:
Some of our test team are having issues with the splash animation persisting

Running master from this morning.
Ubuntu 16.04
Chrome: Version 81.0.4044.113 (Official Build) (64-bit)

[image]
[image]

janpaul123, 2020-04-29:
Oof oh no! Any errors either in the browser console or in the top bar?

jasonimercer, 2020-04-29:
The console is showing

webpack-internal:///./node_modules/react-dom/cjs/react-dom.development.js:558 Warning: Cannot update during an existing state transition (such as within `render`). Render methods should be a pure function of props and state.

jasonimercer, 2020-04-29:
Seems 2 chrome extensions were running: LastPass and Toby. With both disabled the issue went away, I'll work on getting the steps to reproduce on a fresh machine, forward the info along and you can decide if it's something that should be fixed, documented or ignored for now.

janpaul123, 2020-04-29:
Ooh funky, yeah browser extensions can do strange things. Also be sure to check out #407 — we'll merge that soon once we've fixed the screenshot tests, but it contains a bunch of bugfixes and such.

jasonimercer, 2020-04-29:
OK, I couldn't reproduce the issue reported. We'll add the reminder about extensions to our internal pages for WebViz.

Since I can't reproduce and we have a workaround for the people who reported this, I'm happy to have this closed and chalk it up to "spooky extension stuff".

jasonimercer, 2020-04-29:
407 Looks great! Can't wait to update :)

janpaul123, 2020-04-29:
Sounds good; I'll close this out then!

closed 2020-04-29 by janpaul123