cruise-automation/webviz #421: Load RobotModel into ROS 3D panel

BrodieBirko, 2020-05-08:
I am wanting to view my RobotModel in webviz and have the models correspond to joints like they do using urdf's in RVIZ. Is there currently any support for something similar to this? 

There is a [Cruise video](https://youtu.be/_3_Bb-dlq0Q?t=1143) showing a ROS 3D panel with a car model and pointcloud/mapping data, this is in the general direction I want to be heading. Could you share any information on how this was achieved?

zwmisdu, 2020-05-08:
Similarly, I am looking forward to the result of 3d presentation in the video ! And i  notice that the author use a bag named cruise.bag, how could we get it ? About the way to realize the 3d result, I believe that we should learn sth about WORLDVIEW.
Now, I want to get the cruise.bag to learn more in detail.

janpaul123, 2020-05-08:
@BrodieBirko unfortunately the Cruise car model is currently hardcoded into Webviz; we don't currently support loading arbitrary models.

@zwmisdu That's a proprietary Cruise bag which we can't share, but we have a Webviz demo here: https://webviz.io/app/?demo which uses this bag: https://open-source-webviz-ui.s3.amazonaws.com/demo.bag

BrodieBirko, 2020-05-10:
Sweet, that answers my questions.

closed 2020-05-10 by BrodieBirko

srsidd, 2020-06-01:
Is this feature on the roadmap? (Loading any robot model from the urdf?)

janpaul123, 2020-06-01:
Currently not, but happy to take contributions!