cruise-automation/webviz #422: About the dataset cruise.bag!

zwmisdu, 2020-05-08:
Hey, I really appreciate your excellent project.From youtube and ROScon2019, I have both viewed your demo during the presentation.And I can use webviz on my device and see the same result by play demo.bag. For further, I would like to realize the 3d rendering just like you showed in the demo :the car is a 3d model and the road is more like the real environment and so on. I notice that you used a bag named cruise.bag, so how could I get that bag to help my research? Thank you.

janpaul123, 2020-05-08:
Hi, I saw you also posted a comment here about the bag file which I answered: https://github.com/cruise-automation/webviz/issues/421#issuecomment-626001381

For the map you can use nav_msgs/OccupancyGrid; see e.g. how that works in RViz here: http://wiki.ros.org/rviz/DisplayTypes/Map (it works similarly in Webviz).

If you check out the repo and then run `npm run bootstrap && npm run build && npm run docs` you should get a local server. We currently don't have a command for building a static page, but we're tracking that here #406 and I have some suggestions there for implementing it if you want to give it a shot; shouldn't be too difficult. Alternatively you can also directly host the code from the gh-pages branch: https://github.com/cruise-automation/webviz/tree/gh-pages

zwmisdu, 2020-05-09:
> Hi, I saw you also posted a comment here about the bag file which I answered: [#421 (comment)](https://github.com/cruise-automation/webviz/issues/421#issuecomment-626001381)
> 
> For the map you can use nav_msgs/OccupancyGrid; see e.g. how that works in RViz here: http://wiki.ros.org/rviz/DisplayTypes/Map (it works similarly in Webviz).
> 
> If you check out the repo and then run `npm run bootstrap && npm run build && npm run docs` you should get a local server. We currently don't have a command for building a static page, but we're tracking that here #406 and I have some suggestions there for implementing it if you want to give it a shot; shouldn't be too difficult. Alternatively you can also directly host the code from the gh-pages branch: https://github.com/cruise-automation/webviz/tree/gh-pages

Thanks for your quick reply.I have used Rviz expertly so I know how to display the map on it and also have successfully run a local server with my own data. Above all, I think that the map is too plain to show it to others. From your video at ROScon2019, I find a fancy map  which is quite similar with the realworld, we can see the road and zebra stripes aside with the 3d model of the car.And from the previous question, you said the car was hardcoded. So may you give me some adive about how to create the map like the realworld and how to add a car model(maybe it is not supported?)
Thanks again!

janpaul123, 2020-05-09:
Our map was constructed from lidar data by special mapping vehicles, and is currently proprietary, unfortunately. It would be cool to one day support loading map tiles from Mapbox or so though!

Adding your own car model is currently not supported, though if you want to fork Webviz yourself you could replace our car model in this directory: https://github.com/cruise-automation/webviz/tree/1c5cfa783eab39f237c59054d0dc899aee58228a/packages/webviz-core/src/panels/ThreeDimensionalViz/commands/CarModel and then publish a Pose, and then in the settings for that Pose topic enable the car model. But yeah, that's less than ideal.

It's a good feature request though to formally support different 3d models at some point!

zwmisdu, 2020-05-09:
> Our map was constructed from lidar data by special mapping vehicles, and is currently proprietary, unfortunately. It would be cool to one day support loading map tiles from Mapbox or so though! [...]

Okay, get it! Thanks again.

zwmisdu, 2020-05-09:
[image]

zwmisdu, 2020-05-09:
> Our map was constructed from lidar data by special mapping vehicles, and is currently proprietary, unfortunately. It would be cool to one day support loading map tiles from Mapbox or so though! [...]
[image]

Hey! I just watch the video on youtube [https://www.youtube.com/watch?v=_3_Bb-dlq0Q](url)agian. And when you show the demo at 17:50 in the video, I see several options on the  top left corner of the page  just like the topic to be chosen in Rviz ,including Map、Car、Localization ……
So, I wonder 
1)Is the **Map** **rendered** based on the  occupancied grid map with WORLDVIEW? I haven't figured out the relationship between webviz and worldview yet.I just want to create a simliar map of the area of a park, not too large, and I know how to create the map with lidar data but the map is not so fancy as yours.
2)the option **Car**is binding with the pose you mentioned in the previous answer? I try to make clear your reply ' then in the settings for that Pose topic **enable** the car model'How to enable the model?
3)I am really interested in those options on the  top left corner of the page(shown in picture uploaded),for example when you choose Lidar,the other objects are displayed separately, so the Lidar would be the processed information based on raw data from lidar?

zwmisdu, 2020-05-11:
@janpaul123

janpaul123, 2020-05-11:
Ah in the open source version we don't have those categorizations currently, but we render topics in the same way there. So just enable the topics that you want to see. In the future we'd like to make it so you can make your own categorizations like "Lidar" and such.

And in our internal version we render the map in a different way (using a tile server), but you should be able to get a similar effect using the occupancy grid.

When you have a `geometry_msgs/PoseStamped` message you can turn on our car model like this:

[image]

janpaul123, 2020-06-09:
Btw I set up a Slack workspace to make it easier for us to chat; join us using the link here: #461