cruise-automation/webviz #423: webviz won't connect to robot?

bumblebeerobotics, 2020-05-09:
Hello I'm, trying to use WebViz for a project and I'm not sure what is going on/why it is not working. 

I have a remote websocket running for my ROS robot at ws://192.168.1.24:9090 and I changed the ports and URL and everything so I ran https://webviz.io/app/?rosbridge-websocket-url=ws://192.168.1.24:9090/ in the web browser. 

And I went to google chrome settings and changed the settings to allow insecure connections. What am I doing wrong here? Why can't I get webviz to connect to a remote websocket?

jtbandes, 2020-05-09:
What errors are you seeing?

bumblebeerobotics, 2020-05-09:
I'm not seeing any errors which is why I'm confused. When I tried to connect to this web-socket using ROSWEB it connected fine and I was able to get data out of the system. And then if I look on my robot itself the rosbridge outputs to /rosout "one client connected." or something similar

I went to webviz and put in the same ws url, and /rosout showed no message that there was a client connected?

bumblebeerobotics, 2020-05-09:
For reference: this works: https://github.com/EESC-LabRoM/rosweb

bumblebeerobotics, 2020-05-10:
For a little more information: I brought up a Raw Messages panel on the webviz and typed /joy to see the raw messages for /joy that's what I started with trying (my bigger issue is that I don't see the client connected though as I described above)

janpaul123, 2020-05-11:
What version of `rosbridge_server` are you running?

bumblebeerobotics, 2020-05-11:
I just went here: http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge and did sudo apt install ros-melodic-rosbridge-suite ... does that give me the latest version? How would I check?

janpaul123, 2020-05-11:
You can check by running `apt policy ros-melodic-rosbridge-server`. Which version gets installed when you run `apt install` depends on a bunch of things.

bumblebeerobotics, 2020-05-11:
[image]

Here is the version (I think it's 0.11.5-1bionic.20200413.152945). Thanks for the info :)

janpaul123, 2020-05-11:
Hm, that version should be fine. 0.11.6 is the latest, but 0.11.5 should work. Could you maybe record a video of what's happening? If we can't figure it out from that, I might be able to set you up with a screenshare with someone on our team.

bumblebeerobotics, 2020-05-12:
Hey! Here's a link to the video I recorded, I'm ssh'ed into the robot from VS Code and then the google Chrome is on my macbook. 

https://www.adim.io/post/chip-updates-websocket-won-t-work-updates

bumblebeerobotics, 2020-05-12:
let me know if you need more information?

janpaul123, 2020-05-12:
Hm not sure, can you email me at jp.posma@getcruise.com and we can do a screen share.

flynneva, 2020-05-12:
try passing the devices IP address to the rosbridge_webserver. it looks like right now you're just [passing in the default which is `localhost`](https://answers.ros.org/question/250959/connecting-different-computers-via-rosbridge/).

also, look into these other topics which seem similiar:
- [wss support](https://github.com/RobotWebTools/rosbridge_suite/issues/232)
- [wss support 2](https://github.com/RobotWebTools/rosbridge_suite/pull/27)
- [wss support 3](https://answers.ros.org/question/207068/rosbridge-cannot-connect-to-secure-websocket-timed-out-error/)

bumblebeerobotics, 2020-05-12:
I'll definitely try adding the IP to rosbridge_webserver, but I also enabled "use unsafe scripts" for webviz on chrome will that not fix the security issue with https?

bumblebeerobotics, 2020-05-12:
Are you saying I need to do : https://answers.ros.org/question/198647/rosbridge-websocket-ssl-connection-aborted/ kind of stuff to setup ssl?

flynneva, 2020-05-12:
my comment above was just providing some comments on where I think the issue is. since all we have is just a video to debug from and its not entirely clear what the issue is, I cant really say what will solve your problem.

bumblebeerobotics, 2020-05-12:
thanks for the info :) I was just wondering if saying my robot was running a jetson tx1 does that change anything? realized I haven't mentioned that and that I know they're kinda weird.

flynneva, 2020-05-12:
a good way to test if this is the issue is to try webviz from the device directly from the device.
can you connect the jetson tx1 to a monitor and open a browser with webviz there?

flynneva, 2020-05-12:
I know this kind of defeats the purpose of what you are trying to do...but a good debug step would be to test webviz locally on the jetson tx1 (if you can hook it up to a monitor & get a browser open). if that works, then it most likely is an issue with the security (https vs http) as linked above.

bumblebeerobotics, 2020-05-12:
Ah. This was a good suggestion! I did hook it up and run a local test and it does work locally so I think you're right on the ssl permissions side. I'm speaking with JP in a few minutes and I think we'll come up with a solve - if not I'll re-post here if we do, I'll close the issue :) Thanks so much for your help.

bumblebeerobotics, 2020-05-12:
JP called me and we fixed the problem - thank you so much t of both of you - it was lovely meeting you guys!

closed 2020-05-12 by bumblebeerobotics

janpaul123, 2020-05-12:
For future reference, we had to do this:

[image]

A modal with this image + some instructions should have been shown though, so that's a bug that I'll have to look into. Nice meeting you too Adi!

jaguardo, 2020-12-09:
I'm not seeing that option on any of the browsers I've tried... is there a browser setting to change?

jaguardo, 2020-12-10:
So for those that are interested, for Chrome I had to do a few extra steps... I wasn't getting the "sheild" and at first I was getting the "Secure" symbol on Chrome:  
[image]
even when I added the ip address to another computer on the network. makes it appear everything is ok... unfortunately it will not load!! with no warning on why...  
[image]
so, by clicking on the "lock"
[image]
and opening site settings:
[image]
I can set chrome for that website to allow insecure content:
[image]
you can now reload the page (Chrome will pimp you to do this anyways)
now, when you load the basic website:
```
https://webviz.io/app/
```
you will still see the traditional lock, but when you use the alternative address for an ip, you will not get the lock, but a nice "not secure" warning and the shield as you would in previous editions of chrome?
[image]
now by clicking on the shield you can load those scripts:
[image]
hope it helps someone!

vidaaudrey, 2020-12-10:
Thank you @jaguardo  for documenting the details.

marcelino-pensa, 2020-12-29:
Would it be possible to add a link to this issue in the error message that `webviz.io/app` shows up? I spent quite a bit of time trying to figure out why my `shield` wasn't showing up, which could be easily solved by visiting this page (I had to google very specifically to end up in here)

cjds, 2021-04-09:
@vidaaudrey do you mind if I make a PR that updates the help?

vidaaudrey, 2021-04-11:
@cjds  thank you! Please feel free to make a PR for this and tag me for reviewing :)