cruise-automation/webviz #427: [feature request] turn webviz-core into a react component library

flynneva, 2020-05-12:
hey webviz!

I've been diving into how to add ROS connectivity to [my personal website](https://flynnlabs.dev) and I keep coming back to this idea of turning the webviz-core package into its own react component library. I think this is something that a lot of people would love to do and relates sort of to other requests of building their own 'webviz-like' site. I've gone ahead and made a simple one called ['react-ros'](https://www.npmjs.com/package/react-ros) that anyone can `npm install` and get going hooking ROS into their own website.

I'm just now starting to learn how to create my own react component libraries and still trying to understand how all the components would all work together. With webviz-core, the tricky part I think will be the hooks that have been developed to handle the ROS connection and passing it to each component automatically. I've tried to read through the webviz source code but usually get hungup inside the `loadwebviz()` function since this is where all the hooks are loaded in I think.

I'd be interested to hear your thoughts on this and how difficult a project like this would be.

As always, love the webviz project and the work you guys are doing!
-Evan

janpaul123, 2020-05-12:
Yeah this is a long time dream, but it would be a ton of work, plus it would make it much harder for us to keep developing Webviz because we'd have to be extremely careful about changes potentially breaking other projects. And we still have a number of big architectural changes that we want to make that would be really slowed down by that. So we don't want to do this right now, but if you want to make a fork of Webviz and do it yourself then go for it!

Note that we have already done this for the few components that we think are stable enough: [Worldview](https://github.com/cruise-automation/webviz/tree/master/packages/regl-worldview), [rosbag.js](https://github.com/cruise-automation/rosbag.js), and some smaller ones.

flynneva, 2020-05-12:
what big architectural changes are you talking about? :) interested to learn where webviz is heading!

I really have hardly any experience with react so i'll just keep poking around and learning. Most likely by the time I'd learn enough to try and do something like this on my own you guys will have already done it.

janpaul123, 2020-05-12:
We want panels to be able to show buffered data without being tied to the current playback position. E.g. you could imagine the Plot panel just showing the entire bag's worth of data, instead of accumulating when playing back. Or in the 3d panel you might want to plot some topics into the future, and then scrub back and forth, etc.

Also, we want to be able to gather messages over the websocket in a buffer to allow seeking back in time, like TiVo for your robots.

Those are some big changes in how data gets consumed by panels. We're also thinking about rewriting substantial parts of our application in WebAssembly for better performance.