cruise-automation/webviz #430: [feature request] IMU Visualization using sensor_msgs.imu

bumblebeerobotics, 2020-05-13:
Hey all - just submitting this for tracking purposes. Was playing with WebViz today and was looking at a new feature idea:

"See most legged robot and aerial robots have the long-range wifi data transmission and stuff to be able to use webviz as a platform for debugging, but both of these platforms rely heavily on IMUs and specifically the orientation of the IMU for control. I think a feature (maybe part of the 3D block) that takes in a standard ROS IMU message and just produces some rectangle or polygon that can help you visualize the IMU orientation would be really helpful to developers in aerial and legged robots!

Just something very simple, it takes in the standard IMU message from ROS, calculate the Euler angles, and outputs a figure that shows the block and the roll/pitch/yaw (see the gif below)."

[image]

janpaul123, 2020-05-13:
Great idea! If anyone wants to take this on, this should be relatively straightforward to add by doing something similar to our support for the Pose message here: https://github.com/cruise-automation/webviz/blob/deae0a7bced6829c55682354c9f40fc2cb79da28/packages/webviz-core/src/panels/ThreeDimensionalViz/SceneBuilder/index.js#L457-L460

bumblebeerobotics, 2020-05-13:
To whoever does end up working on this :) - I did look around at some things on visualizing came up with this:

[image]

So for this, I made a ROS node that takes in the IMU data and re-publishes another /tf between the base_link of the robot and the /imu_frame by passing the imu outputted quaternion to the tf itself. 

Then I made a block using ros markers (a cube) that just followed the same imu_frame or used the same translation from the base_frame: https://www.adim.io/post/chip-updates-tf-s-and-music-notes-updates some details. 

So this approach worked but a better thing to do might be the 3D viewer recognizes the imu message itself, and visualizes that without having to setup any transforms.

bumblebeerobotics, 2020-05-13:
this is kinda weird because usually base_link would move with the IMU it wouldn't stay fixed...