cruise-automation/webviz #431: [feature request] GPS Data Visualization on Map using ~sensor_msgs/NavSatFix

bumblebeerobotics, 2020-05-13:
Hey all - another suggested feature for you: 

Something cool on some of the old web visualization tools for ROS was the ability to select a sensor_msgs/NavSatFix message (GPS global coordinates) and visualize that on a map background. Something like that could be good for debugging GPS data and also, even, if you want to make a custom "Ground Control Station" or something. 

http://wiki.ros.org/gpsd_viewer --> ROS package for viewing GPS data!

janpaul123, 2020-05-13:
Good idea. Seems best to put it in the 3d panel, though we'd have to figure out how to pull in map tiles in the right coordinate frame and not when using a local coordinate frame.

luqnaim, 2021-11-02:
Bump. Would be super cool to have this!

jtbandes, 2021-11-02:
HI folks, for anyone interested in this feature, we have a Map panel in Foxglove Studio: https://foxglove.dev/docs/panels/map