cruise-automation/webviz #436: How to add custom Image command into ThreeDimensionalViz?

xiaodongzhao, 2020-05-18:
Hi, 
Is it possible to add a custom command into ThreeDimensionalViz? 
I would like to display map images in ThreeDimensionalViz, for example, using the code from 
https://github.com/cruise-automation/webviz/issues/225. 

Is it possible to do so? what are the steps needed? Thanks

janpaul123, 2020-05-18:
This is not officially supported, for maps I'd recommend using [OccupancyGrid](http://docs.ros.org/api/nav_msgs/html/msg/OccupancyGrid.html). We'd like to support this in the future though. But of course you can always fork our repo and attempt to do it manually, but that's not straightforward.

vidaaudrey, 2020-05-18:
If you have some marker format that are not currently supported, you could add it to `renderAdditionalMarkers` hook and apply your custom command:

https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/World.js#L128-L130

xiaodongzhao, 2020-05-18:
> This is not officially supported, for maps I'd recommend using [OccupancyGrid](http://docs.ros.org/api/nav_msgs/html/msg/OccupancyGrid.html). We'd like to support this in the future though. But of course you can always fork our repo and attempt to do it manually, but that's not straightforward.

Thanks @janpaul123 and @vidaaudrey , OccupancyGrid is a good solution, however we need to display coloured image maps. Do you have other suggestions for displaying colour images? I tried using PointCloud as well, but it didn’t scale when zoomed.

janpaul123, 2020-05-18:
That's a great idea. I don't think we currently support that. Our reference implementation for the 3d panel is RViz, do you happen to know if this is possible there? I could only find it as a custom plugin: https://github.com/lucasw/rviz_textured_quads/tree/image_topic and it's not entirely clear what kind of message definition they use.

xiaodongzhao, 2020-05-19:
It seems https://github.com/lucasw/rviz_textured_quads/tree/image_topic is using Image and TF frame, while the original (https://github.com/MohitShridhar/rviz_textured_quads) is using a customized message. I don't think RViz supports display images in the 3D scene.

janpaul123, 2020-05-19:
> Image and TF frame

janpaul123, 2020-05-19:
> Image and TF frame

Interesting; yeah you can position the image using the `frame_id`, but I don't think there's enough information in those messages right now to know what the width and height in 3d-space should be? I wonder how they do that; maybe they let you just manually configure that? A customized message might be better if you want to control those dimensions programatically.

In any case, we don't have plans to implement this right now, but we'd happily take a PR if you want to give it a shot.

janpaul123, 2020-06-09:
Btw I set up a Slack workspace to make it easier for us to chat; join us using the link here: #461