cruise-automation/webviz #440: How to use 2DPlot panel

azuremeteor, 2020-05-26:
I have a set of coordinate point X, Y point want to plot on graph so i tried to use 2DPlot panel.
I saw the example require a message type is webviz_msgs/2DPlotMsg. But I cannot found this msg type, so I created a package call webviz_msgs and a msg type 2DPlotMsg in python.
But due to "2DPlotMsg" is start with a number so I cannot import it other script and rosbrige server to use.

May I know how can I use 2D plot? Because my coordinate update rate is 60Hz, using the plot function will cause to much point in the same graph and the performance will become low.

Thx

esthersweon, 2020-05-26:
Hey there, 

The datatype field is actually not required - it is specified in the docs, just to denote that the message must be in a specific format to be consumed by the 2D Plot. 

As long as your message looks something like the example message below, it should be able to be visualized in the panel.

```js
{
  op: "message",
  topic: "/plot_a",
  datatype: "some_datatype",
  receiveTime: { sec: 1532375120, nsec: 317760607 },
  message: {
    title: "This is Plot A",
    xAxisLabel: "This is my X axis label",
    yAxisLabel: "This is my Y axis label",
    lines: [
      // 1 solid red line
      { label: "solid-line", borderColor: "red", backgroundColor: "red", data: [{ x: 0, y: 0 }, { x: 5, y: 5 }] },
      // 1 dashed pink line
      {
        label: "dashed-line",
        borderDash: [5, 5],
        borderColor: "pink",
        backgroundColor: "pink",
        data: [{ x: 1, y: 0.5 }, { x: 5, y: 3.5 }],
      },
    ],
    points: [
      // 3 circular blue points
      {
        label: "circle-point",
        pointBackgroundColor: "blue",
        data: [{ x: 1.5, y: 2.5 }, { x: 3, y: 4 }, { x: 4, y: 3.5 }],
      },
      // 2 teal Xs
      {
        label: "cross-point",
        pointBackgroundColor: "teal",
        pointBorderColor: "teal",
        pointStyle: "crossRot",
        pointRadius: 10,
        data: [{ x: 2, y: 1 }, { x: 4, y: 1 }],
      },
    ],
    polygons: [
      // 1 light gray triangle
      {
        label: "polygon",
        borderColor: "lightgray",
        backgroundColor: "lightgray",
        data: [{ x: 3, y: 1 }, { x: 4, y: 2 }, { x: 4.5, y: 1.5 }],
      },
    ],
  },
}
```

janpaul123, 2020-06-09:
Btw I set up a Slack workspace to make it easier for us to chat; join us using the link here: #461

closed 2020-10-19 by vidaaudrey