cruise-automation/webviz #442: Error when conneting to rosbridge after running static-webviz

Shiming94, 2020-05-26:
After building the dependencies and running `npm run serve-static-webviz`, then typing http://localhost:8080/ in Chrome, it can directly lead to the application page, rather than run http://localhost:8080/app one more time as before. But if I run rosbridge afterwards, the application page will throw an error out. The details are shown in the following screenshot.
[image]
Does somebody know, what is the reason for the error, and how can I solve it?
Thanks in advance.

janpaul123, 2020-05-27:
Hm, interesting, I can't reproduce. Do you get the same error if you use `docker run -p 8080:8080 cruise/webviz` instead of `npm run serve-static-webviz`?

I think I've seen this error before though, but I forgot in which context. I'll do a release from our internal repo to see if that helps.

janpaul123, 2020-05-27:
Ok I updated `master` with our latest code changes (https://github.com/cruise-automation/webviz/pull/443), can you pull and try again and see what happens?

Shiming94, 2020-05-28:
Hi, thanks a lot. I have tried the newest code. But it didn't work The error is same. At first I ran `docker run -p 8080:8080 cruise/webviz`, it will work. But if I ran `npm run serve-static-webviz` once, the localhost:8080 is occupied, and docker will also not work.

janpaul123, 2020-05-28:
Yeah you can't run them at the same time. But so you're saying that the Docker version did work? What about https://webviz.io/app, does that work? If you're running the new code, can you try checking out a fresh repo (throw away the old repo), and going through all these steps again: https://github.com/cruise-automation/webviz#building-manually

janpaul123, 2020-06-02:
We think we've identified the issue and will be issuing a patch soon. The issue is in the layout migration code. For now can you run `localStorage.clear()` in the browser console, to clear out your layout, and tell me if that helped?

Shiming94, 2020-06-03:
Hi Jan, I have pulled the latest webviz from github and it seems that the connection problem is solved. Thanks a lot. And then I meet another problem: in the 3D Panel I want to visualize a message in type PoseStamp, at first the message is visualized as a cylinder, then I choose "Use 3D Model" and the message disappears. Do you know what is the reason of it?

janpaul123, 2020-06-03:
Hm, for me that seems to work:

[image]

However, be aware that that option currently only supports setting a Cruise car, so it's probably not very useful for you. We only really support the cylinder right now, though you can modify the dimensions.

chenguoxi1987, 2020-06-08:
Hi,I still have an error as below when I run "docker run -p 8080:8080 cruise/webviz" and  "roslaunch rosbridge_server rosbridge_websocket.launch“:

[ERROR] [1591610444.208615]: [Client 0] [id: call_service:/rosapi/topics_and_raw_types:1] call_service InvalidServiceException: Service /rosapi/topics_and_raw_types does not exist

[image]

janpaul123, 2020-06-08:
Ah, that usually means that you're running an old version of the `rosbridge_server`. Can you check which version you're running? We require 0.11.4 or later.

janpaul123, 2020-06-09:
Btw I set up a Slack workspace to make it easier for us to chat;  email me at jp.posma@getcruise.com and I'll invite you 😄

chenguoxi1987, 2020-06-09:
> Ah, that usually means that you're running an old version of the `rosbridge_server`. Can you check which version you're running? We require 0.11.4 or later.

I check my rosbridge_server veision, and it's 0.11.8.

chenguoxi1987, 2020-06-09:
> Ah, that usually means that you're running an old version of the `rosbridge_server`. Can you check which version you're running? We require 0.11.4 or later.

I updata my rosbrige_server version from 0.11.3 to 0.11.8, but I receive the same error.
[Client 1] [id: call_service:/rosapi/topics_and_raw_types:266] call_service InvalidServiceException: Service /rosapi/topics_and_raw_types does not exist

Xia0ben, 2020-06-23:
Hello there,
Was trying to get webviz up and runnin' on our robot today, but got the same error on the rosbridge end :

janpaul123, 2020-06-23:
@Xia0ben we chatted more in Slack, and it looks like perhaps something went wrong in the upgrade, because on another computer @chenguoxi1987 did get it to work. So I'd still recommend upgrading to 0.11.8.

I'll add a note to the readme about the version that we test against!

Xia0ben, 2020-06-24:
Hi !
Thanks for the fast reply, I'll give it a spin probably next week !

jbcpollak, 2022-05-17:
sorry to revive a dead thread but I am seeing this issue today with Melodic:

> [ERROR] [1652746592.655572]: [Client 0] [id: call_service:/rosapi/topics_and_raw_types:979] call_service InvalidServiceException: Service /rosapi/topics_and_raw_types does not exist

```
rosuser@host:~$ rosversion rosbridge_server
0.11.13
```