cruise-automation/webviz #444: unable to show tf tree in 3D panel.

ZhengPan2014, 2020-05-28:
as showed below, right part is rqt tf which indicates tf is good, left part is webviz, tf is not available, How to fix it? 
[image]

janpaul123, 2020-05-28:
Do you have a bag that you can share for us to reproduce?

@vidaaudrey @davidswinegar maybe this is a bug with the new topic picker, can you follow up when we have a bag?

vidaaudrey, 2020-05-28:
Will do once @ZhengPan2014  have a bag for us to test.

ZhengPan2014, 2020-05-29:
> Will do once @ZhengPan2014 have a bag for us to test.

Yes, bag is available [here](https://pan.baidu.com/s/1zy6joV5yTtPYz3EAhv3dKw), access code is: xrx6.

vidaaudrey, 2020-05-29:
Looks like a bug. Will investigate and try to fix today.

vidaaudrey, 2020-05-30:
@ZhengPan2014 The problem is fixed now. Please let us know if you see any other issues. 
[image]

closed 2020-05-30 by vidaaudrey