cruise-automation/webviz #461: Slack invite link

janpaul123, 2020-06-09:
Since Slack invite links expire after 30 days, I figured we can just post them here whenever it's needed.

janpaul123, 2020-06-09:
https://join.slack.com/t/webvizchat/shared_invite/zt-evd6yzee-MmfUfe_hRvf~t9HBPpKkdQ

closed 2020-06-09 by janpaul123

maxlein, 2020-09-02:
Pls update the invite

troygibb, 2020-09-02:
Here you go! https://join.slack.com/t/webvizchat/shared_invite/zt-h036wkmu-y5AxRCzSRdNbAJw0zYMA4g

lukicdarkoo, 2020-10-05:
Please update the link

troygibb, 2020-10-05:
Done! https://join.slack.com/t/webvizchat/shared_invite/zt-hyo9jggj-n4dwegVa290HM5PpPRWBZw

rgov, 2020-11-10:
@troygibb A new link please?

troygibb, 2020-11-10:
Done again! https://join.slack.com/t/webvizchat/shared_invite/zt-j6z2kcbl-aJHPkK~99~oYHv7LWrODRA

CalvinKroese, 2020-12-11:
@troygibb Could you send a new link please?

troygibb, 2020-12-11:
Done again again! https://join.slack.com/t/webvizchat/shared_invite/zt-k3nj8jx0-qhg~wTmqfklJWVjPRWiVBQ

ammaryasirnaich, 2021-01-11:
Please can you send the invitation link !

inactivist, 2021-01-20:
Please provide a new invitation link.  The most recent one has expired.

[image]

troygibb, 2021-01-20:
here you go! https://join.slack.com/t/webvizchat/shared_invite/zt-l7x2ocqc-2uuhztdaOv50GtzTQh8jLg

yashvakil, 2021-02-23:
The link has expired could you share another

troygibb, 2021-02-23:
Done. https://join.slack.com/t/webvizchat/shared_invite/zt-mrp65eb6-ZOrtb1qXrofFRXqYBvpyXw

shreyas2311, 2021-04-01:
The link has expired. Can you share another? thanks

troygibb, 2021-04-01:
https://join.slack.com/t/webvizchat/shared_invite/zt-ou31t3sg-wyaa0fpTby1KvJXGbXPuvg

m2themax, 2021-05-03:
Link has expired. Can you please post a new one. Thanks!

troygibb, 2021-05-03:
https://join.slack.com/t/webvizchat/shared_invite/zt-q58hdnw0-HeV8QkOumMQDrBgtzm5jYg

thewalk, 2021-06-07:
please share a new link, thanks!

troygibb, 2021-06-07:
https://join.slack.com/t/webvizchat/shared_invite/zt-r6f2atsn-5yYHSG7JaPqtbMJUbatpaA

nonlinear1, 2021-07-14:
This link is no longer active. Could you send a new link?

troygibb, 2021-07-14:
https://join.slack.com/t/webvizchat/shared_invite/zt-t0xhhiur-VZZrViVqbYLOwWpYJB9PZw

stefan-pdx, 2021-08-16:
Hi @troygibb. Would you mind updating the link please? Thanks!

ShepelIlya, 2021-08-29:
@troygibb Could you post a new link please?

troygibb, 2021-08-30:
https://join.slack.com/t/webvizchat/shared_invite/zt-um7wbhu9-xaMavlo6AVwoWZsTaa7Upg

rfsch, 2021-09-21:
@troygibb Could you post a new link please? Thanks!

Wudadad, 2021-09-24:
> @troygibb Could you post a new link please? Thanks!

+1

troygibb, 2021-09-24:
sorry for the late reply! https://join.slack.com/t/webvizchat/shared_invite/zt-wfu6cctt-k5L2dN6XoZWbX_ULm8QFGQ

jtbandes, 2021-09-29:
FYI, we also have a Slack for [Foxglove Studio](https://github.com/foxglove/studio) (Webviz fork) here: https://foxglove.dev/slack

AnExplorer28, 2021-11-03:
Hi, the Previous link is no longer active. Can you please share the new link? Thanks!

troygibb, 2021-11-03:
https://join.slack.com/t/webvizchat/shared_invite/zt-yjc7katc-iqH1Lc0b1Tmm6LxiuPFvkQ

Beaudidly, 2021-12-31:
Hey! Could I please have an updated invite please?

troygibb, 2022-01-03:
https://join.slack.com/t/webvizchat/shared_invite/zt-112b2nhdg-wmyG19eD03hfymkMKIsFsA

vicente879, 2022-02-24:
Hi, the current link has expired. Could you post an updated invite please? thanks!

C1oudust, 2022-02-25:
Hey! Could I please have an updated invite please?

teknetik, 2022-06-10:
Please could we have an updated invite?

rsinnet, 2022-07-11:
Updated invite please?

LJL0511, 2022-09-22:
> Could you send a new link please?

amacneil, 2022-09-22:
Most of the former Webviz chat has moved to the Foxglove Slack channel, you can join here: https://foxglove.dev/slack.