cruise-automation/webviz #466: How to connect robot in Gazebo to WebViz

sneakyc4k3, 2020-06-17:
Could I connect a virtual robot such as the turtlebot in Gazebo to WebViz using WebSocketServer? If yes, how?

janpaul123, 2020-06-17:
Yes, if you can get a rosbridge_server to work, then you can connect to it with Webviz. I don't know how to set that up with Gazebo though, might be worth asking the [rosbridge people](https://github.com/RobotWebTools/rosbridge_suite/). Also feel free to join our [Slack channel](https://github.com/cruise-automation/webviz#slack).

sneakyc4k3, 2020-06-18:
Yes, I know how to connect Gazebo to rosbridge. However I do not know how to use Webviz to connect Webviz to rosbridge_server.

janpaul123, 2020-06-18:
Ah — we have a help page that is accessible in the app by clicking on the question mark in the top right. It explains a bit on how to configure the Websocket URL:

> By default we try connect to a [rosbridge_server](http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge) using a WebSocket on `ws://localhost:9090`. If you want to use a different URL, use the `?rosbridge-websocket-url=…` parameter. For example, to connect to `ws://localhost:4321`, use [this URL](https://webviz.io/app?rosbridge-websocket-url=ws://localhost:4321). If you're using a more complex URL, it'd be good to [url-encode](https://www.urlencoder.io/) it, for example [like this](https://webviz.io/app?rosbridge-websocket-url=ws%3A%2F%2Flocalhost%3A4321). This also applies to other parameters mentioned below.

So yeah, typically you can take a URL like this: `https://webviz.io/app/?rosbridge-websocket-url=ws://localhost:4321` and replace `localhost` by the hostname or IP address of your ROS machine (if it's a remote one), and `4321` by the port number that you configured in rosbridge.

janpaul123, 2020-06-18:
By the way, it might be necessary to click on a "Load unsafe scripts" button if the websocket server is in non-SSL mode. We should show a popup in that case that has this image in it:

[image]

But I've seen a bug before where that popup doesn't show up, so if that's the case let me know because I haven't been able to reproduce that, so your input would be helpful there.

sneakyc4k3, 2020-06-28:
Hi- I've managed to connect the websocket to Webviz but nothing is showing up! Also, I'm not receiving the popup.
[image]

janpaul123, 2020-06-29:
Looks like stuff is appearing on /rosout at least. You might want to try opening the Raw Messages panel to look at your topics and if they look like what you expect. The 3d panel only works with RViz markers (and some other messages types like Pose.

sneakyc4k3, 2020-06-30:
How do I add Rviz markers?

janpaul123, 2020-06-30:
I'd recommend getting RViz to work first and to use their guides. Once you have RViz markers working within RViz, you can click in the top left button of a 3d panel and enable those markers. You might also have to click on the little arrow in the right menu to select which TF you're following.

sneakyc4k3, 2020-07-02:
Meaning I have to run Rviz and Gazebo before running WebViz ?

janpaul123, 2020-07-02:
Yeah I'd recommend trying to get it to work with RViz first. Webviz' 3d panel is supposed to be compatible with RViz, so if it works in RViz, then it should work in Webviz. If it then still doesn't work in Webviz, then it might be a bug.

sneakyc4k3, 2020-07-03:
I managed to get the map from Rviz to show in Webviz. Is there a way that the turtlebot could also be displayed there? As right now only the turtlebot's links (different parts) are being displayed.
[image]

janpaul123, 2020-07-06:
Yeah, you'd have to publish an [RViz marker](http://wiki.ros.org/rviz/DisplayTypes/Marker) to represent the turtlebot. We also support `geometry_msgs/Pose`. Here's a current list of the message types that we support in the 3d panel: https://github.com/cruise-automation/webviz/blob/ed49a4c1757c3b6efa8f5d6e862333534453bf61/packages/webviz-core/src/hooksImporter.js#L86

Wudadad, 2021-09-27:
> I managed to get the map from Rviz to show in Webviz. Is there a way that the turtlebot could also be displayed there? As right now only the turtlebot's links (different parts) are being displayed.
> [image]
Hello, I have the same problem, do you have any good solution?