cruise-automation/webviz #469: Can't visualize lidar data on Webviz

rawanbadr, 2020-06-22:
I am running webviz locally and i am playing the rosbag in the background in order to visualize it.The image data is working perfectly after selecting the topic but the lidar data is giving the following error in the picture ,any help please?
[image]

janpaul123, 2020-06-22:
You might have to select a different root TF in the top right.
[image]

Also feel free to join us on Slack: #461.

rawanbadr, 2020-06-23:
i still do have the same issue

vidaaudrey, 2020-06-23:
@rawanbadr Could you record a sample bag and share with us?

janpaul123, 2020-06-23:
@vidaaudrey we're chatting with them on Slack!

castaway2000, 2020-08-03:
I am also having the same issue but with an Ouster Lidar Sensor.

vidaaudrey, 2020-10-19:
@rawanbadr  closing the issue now. Let me know if you still have problem visualizing the topic.

closed 2020-10-19 by vidaaudrey