cruise-automation/webviz #478: http access from remote machines

codegrafix, 2020-07-21:
Hi,

I have been using webviz with Docker for a while now and it works perfectly to visualize all my C++ roses nodes. Thanks a lot for that!

Now I want to give remote access to webviz on my computer from any computer in the local network.
But I could not find any information about how to configure or how to grant permissions to remote computers. 

Could you please share some information about this?
Thank you!

hassoncs, 2020-07-22:
Hi @codegrafix,

I'm glad webviz is working well for you! In order to access webviz from other computers, you might need update the docker configuration in order to forward (or expose) the webviz port (usually using the `-p` flag). You might need to adjust your computer's firewall rules as well. Are you using IP address and port to connect to the computer running webviz?

Here's an article on Docker's networking rules that might be helpful:
https://www.ctl.io/developers/blog/post/docker-networking-rules

```
TL;DR
For reliability, use -p or -P to create specific port binding rules;
...
```

hassoncs, 2020-07-22:
Also, feel free to join our Slack channel to get your questions answered quicker and to chat with other Webviz users: https://github.com/cruise-automation/webviz#slack

codegrafix, 2020-07-22:
Hi @hassoncs,

You were right, I had some problems with the firewall! Thanks for the quick answer.

closed 2020-07-22 by codegrafix