cruise-automation/webviz #485: Support rendering visualization_msgs/ImageMarker messages from Node Playground

jhurliman, 2020-07-25:
Right now, Node Playground can only output to topics with the `/webviz_node/` prefix but the image panel will only seem to render `visualization_msgs/ImageMarker` messages from topics that share a namespace with cameras. I was hoping just setting the `frame_id` of the image marker to the same frame as my camera was enough to make the association.

On a separate note, it's much more difficult to create image markers in Node Playground than 3d markers because there is no ImageMarkerArray message type, so each playground node can only create a single image marker. This is fine for any of the array-compatible primitives like lines and points, but doesn't work out if you want to create more than one polygon for example. This seems to be a ROS limitation though, I'm not sure there's anything webviz can do about it.

troygibb, 2020-07-28:
Topics that share a namespace with the camera are only used to provide the default list. There should be an `Add topic` button at the bottom of that list, which will allow you to subscribe to any topic you like. Is that showing up for you?

As for no `ImageMarkerArray` message type, that's an interesting point. We do support an array type internally. One hack you could potentially use is to change [this line](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/hooksImporter.js#L124) to include `visualization_msgs/MarkerArray`. I believe the types are similar enough where this might work.

jhurliman, 2020-07-28:
@troygibb that's correct, my mouse cursor turns into a circle with a line through it and the greyed out dropdown says "no markers" even though I can see my `visualization_msgs/ImageMarker` messages being continuously published to a `/webviz_node/...` topic with the raw messages panel.