cruise-automation/webviz #490: [information request] lidar point cloud color data

castaway2000, 2020-08-03:
in my implementation it seems that webviz defaults all points to being white. However, in the demo and in my local compiled driver for the lidar sensor i have i can see color coded data when streaming and replaying bag files. How can we visualize the color coded data in webviz?

castaway2000, 2020-08-26:
Reopening as i still have not been able to figure out how to render color based on things like luminosity or distance, etc. Is there a message for the 3d view that i would need to make sure is present? what would I be missing?

flynneva, 2020-08-26:
click `Edit Topic Settings`....from there you should be able to select the color based on field of the pointcloud

[image]

castaway2000, 2020-08-26:
@flynneva is this on all topics or only some topics? i am not sure mine has the edit logo on the standalone, i did not see that till now. That is a big help thank you. feel free to close this out.

flynneva, 2020-08-26:
i think it should be on all of them. its just to edit how the data is visualized so its specific to webviz not the topic.

castaway2000, 2020-08-26:
@flynneva i got to test this a little bit today on my lunch break. turns out if you don't see the pencil next to the logo (this was my case) this means its running default settings. the three dot menu  that you pointed out to select edit settings is sure enough there. that was awesome to flip on. Thanks a bunch.

closed 2020-08-26 by castaway2000