cruise-automation/webviz #491: Marker Array Points Not Scaled Correctly

jasonimercer, 2020-08-04:
Points in a Marker Array message are scaled in units of pixels rather than meters as the unit in other elements.

issue:

https://github.com/cruise-automation/webviz/blob/master/packages/regl-worldview/src/commands/Points.js#L32

vs

code that considers the camera:

https://github.com/cruise-automation/webviz/blob/master/packages/regl-worldview/src/commands/Lines.js#L147-L151

rviz and webviz visualizing points in a marker array:
[image]

example input with points in a marker array 
[2020-07-28-12-26-37.bag.gz](https://github.com/cruise-automation/webviz/files/5022124/2020-07-28-12-26-37.bag.gz)

troygibb, 2020-08-04:
Thanks @jasonimercer for reporting! We may have to refactor our `<Points />` command to account for this issue. 

Can you describe your use case a bit more? Why do you need to express points so exactly? Why not use some of our other primitive shapes?

cc @hhsaez

jasonimercer, 2020-08-04:
We use Marker Arrays for rich diagnostics. Our code publishes lines, polygons, arrows, text, points to give our triage teams an easier view into the state of the robot and the surroundings.

In this marker array we denote remembered obstacles that were point-like as well as obstacles that were identified as moving polygons. We also show demarcations between regions we can observe and will add data to memory, regions we only use to invalidate memory and regions where we rely on memory.

When we are displaying point-like lidar data in our aggregated memory, the points are single pixels using Webviz (which on modern displays and a busy interface, is as good as invisible - see pixel data in screenshot above). We don't need them to be 100% accurate, we just need them to be visible similar to RViz so we can continue to shift the hold-outs towards webviz.

jasonimercer, 2020-08-05:
Callout added to image above demonstrating issue:

[image]

troygibb, 2020-08-06:
Thanks Jason for the context -- I just wrote a ticket to track this issue internally.

jasonimercer, 2020-09-25:
I see code has been merged related to this case. Is it live on the demo site or is there something that needs to be done to trigger it? When I load my example bag file above into the demo site the display appears the same.

troygibb, 2020-09-25:
We're still waiting on some internal updates to be made before updating open source with the new change. @hhsaez can you make sure to update open source once that PR lands?

jasonimercer, 2020-09-25:
Wonderful, happy I wasn't missing something obvious on the demo site :)

hhsaez, 2020-10-19:
@jasonimercer! Webviz has just been updated with the fix for point sizes. Please give it a try when you have the chance.

jasonimercer, 2020-10-19:
Brilliant! I'll try today.

jasonimercer, 2020-10-19:
There may be a factor of 2 issue somewhere? Throwing the test bag above into the live server shows a difference but feels like the squares are about 2x too large?

I'll look closer this evening.

jasonimercer, 2020-10-20:
[image]

Looks like the marker array boxes are now 1m across while the scale calls for a 0.2m x 0.2m boxes.

iwanders, 2020-10-20:
I've filed #529 , which fixes the point scaling with scales below `1.0`.

iwanders, 2020-10-23:
@jasonimercer, I think this issue can be closed now that the scaling is correct in the latest version of `master`?

closed 2020-10-23 by jasonimercer