cruise-automation/webviz #493: npm test fails

jack-slingsby-5, 2020-08-05:
Hiya,
On the most recent version of the master branch the following 3 tests fail: 

```
FAIL  packages/regl-worldview/src/utils/queuePromise.test.js
  ● queuePromise › queues with resolved and rejected promises

    expect(received).toBeUndefined()

    Received: {}

      41 |     expect(calls).toBe(3);
      42 |     expect(callArgs).toEqual([[1, 2], [3, 4], [5, 6]]);
    > 43 |     expect(queuedFn.currentPromise).toBeUndefined();
         |                                     ^
      44 |   });
      45 | 
      46 |   it("returns a promise", async () => {

      at Object.toBeUndefined (packages/regl-worldview/src/utils/queuePromise.test.js:43:37)
```

```
 FAIL  packages/webviz-core/src/util/debouncePromise.test.js
  ● debouncePromise › debounces with resolved and rejected promises

    expect(received).toBeUndefined()

    Received: {}

      32 |     await Promise.resolve();
      33 |     expect(calls).toBe(2);
    > 34 |     expect(debouncedFn.currentPromise).toBeUndefined();
         |                                        ^
      35 | 
      36 |     debouncedFn();
      37 |     expect(calls).toBe(3);

      at Object.toBeUndefined (packages/webviz-core/src/util/debouncePromise.test.js:34:40)
```

```
 FAIL  packages/webviz-core/src/dataProviders/MemoryCacheDataProvider.test.js
  ● MemoryCacheDataProvider › prefetches after the last request

    expect(received).toEqual(expected)

    Expected value to equal:
      [{"blocks": ArrayContaining [], "fullyLoadedFractionRanges": [{"end": 0.04975124378109453, "start": 0}, {"end": 0.8009950248756219, "start": 0.4975124378109453}]}]
    Received:
      [{"blocks": [{"messagesByTopic": {"/foo": [{"message": [], "receiveTime": {"nsec": 0, "sec": 0}, "topic": "/foo"}]}, "sizeInBytes": 600}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, {"messagesByTopic": {"/foo": [{"message": [], "receiveTime": {"nsec": 0, "sec": 10}, "topic": "/foo"}]}, "sizeInBytes": 600}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": [{"message": [], "receiveTime": {"nsec": 0, "sec": 12}, "topic": "/foo"}]}, "sizeInBytes": 600}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": [{"message": [], "receiveTime": {"nsec": 0, "sec": 14}, "topic": "/foo"}]}, "sizeInBytes": 600}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": []}, "sizeInBytes": 0}, {"messagesByTopic": {"/foo": [{"message": [], "receiveTime": {"nsec": 0, "sec": 16}, "topic": "/foo"}]}, "sizeInBytes": 600}, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined, undefined], "fullyLoadedFractionRanges": [{"end": 0.05472636815920398, "start": 0}, {"end": 0.8009950248756219, "start": 0.4975124378109453}]}]
```

jack-slingsby-5, 2020-08-05:
I'm running v12.18.1

troygibb, 2020-08-05:
Try running `v10.16.3` -- we should make this requirement more explicit in our docs.

jack-slingsby-5, 2020-08-06:
Yep that seemed to the the problem! All the tests are passing fine in `v10.16.3`. Thanks :)

closed 2020-08-06 by jack-slingsby-5