cruise-automation/webviz #495: Unable to stream bagfile from presigned s3 bucket

castaway2000, 2020-08-19:
I have a pre-signed s3 url for a bag file. I have tested that it is available, i can download it to my local machine and the curl output shows the appropriate headers present. The Following is a copypaste from my s3 bucket of the CORS configuration and below that is my curl snippet (which does indeed download the file contents). 

**I am attempting to come up with a reason i am still getting the** 
"Error Initializing player" message which then tells me to set the CORS appropriately. 

Thoughts on what could be broken here? 

```
<?xml version="1.0" encoding="UTF-8"?>
<CORSConfiguration xmlns="http://s3.amazonaws.com/doc/2006-03-01/">
<CORSRule>
    <AllowedOrigin>*</AllowedOrigin>
    <AllowedHeader>*</AllowedHeader>
    <AllowedMethod>GET</AllowedMethod>
    <AllowedMethod>HEAD</AllowedMethod>
    <ExposeHeader>Access-Control-Allow-Origin</ExposeHeader>
    <ExposeHeader>ETag</ExposeHeader>
    <ExposeHeader>Content-Type</ExposeHeader>
    <ExposeHeader>Accept-Ranges</ExposeHeader>
    <ExposeHeader>Content-Length</ExposeHeader>
    <ExposeHeader>Range</ExposeHeader>
</CORSRule>
</CORSConfiguration>
```

```
< HTTP/1.1 200 OK
< x-amz-id-2: hueWUxV1sVkyhlm8wCcUJHUBo5BCa51ET2GQTcWL32EhTG9t59KaeSIp7TbGdsGulUM7MpwZM7U=
< x-amz-request-id: E26232BB7590C183
< Date: Wed, 19 Aug 2020 03:15:45 GMT
< Access-Control-Allow-Origin: *
< Access-Control-Allow-Methods: GET, HEAD
< Access-Control-Expose-Headers: Access-Control-Allow-Origin, ETag, Content-Type, Accept-Ranges, Content-Length, Range
< Vary: Origin, Access-Control-Request-Headers, Access-Control-Request-Method
< Last-Modified: Sun, 16 Aug 2020 20:53:26 GMT
< ETag: "f6bbcf5fc47bcd79fc6285eb8256face-33"
< Cache-Control: max-age=86400
< Accept-Ranges: bytes
< Content-Type: application/octet-stream
< Content-Length: 272910573
< Server: AmazonS3
```

castaway2000, 2020-08-19:
Ok for sake of solving my own issue... cause its not very clear in the error documentation it appears that all remote bag urls MUST be url encoded (including private URL's) which the error documentation is a bit misleading. I decided to just encode the url to test it and it solved my problem. 

## As per the docs: 
First of all, be aware that redirects don't work with CORS. So you'll have to point to the resource directly. **So ?remote-bag-url=https%3A%2F%2Fopen-source-webviz-ui.s3.amazonaws.com%2Fdemo.bag will work** (note the URL encoded using "encodeURIComponent"), but pointing to a server that then redirects to a URL like that will NOT work.

If your data is sensitive, **you can generate a signed S3 url, and use that directly** with ?remote-bag-url=. That URL will then only work for a limited time period, and you can have a server that only signs URLs for authenticated users.

**As you can see that makes it sound like if you are a public url you need to encode it, if you are private url you can use it as it is.**

vidaaudrey, 2020-08-20:
Hi @castaway2000, glad you figured it out. Another alternative is to have `remote-bag-url` be the actual file url but use the the BE server to generate a signedUrl for the FE to stream from. Bit work from the BE but that keeps the url short and readable.

castaway2000, 2020-08-20:
@vidaaudrey you will have to excuse me and my ignorance here but what is BE and FE?

vidaaudrey, 2020-08-20:
@castaway2000  sorry for the confusion. The open source version of Webviz doesn't contain any backend, but you could build your own backend API endpoint to take the URL and return the signedURL instead.  e.g. from the [PlayerManager.js](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/components/PlayerManager.js#L233), 
```
const urls = [params.get(REMOTE_BAG_URL_QUERY_KEY), params.get(REMOTE_BAG_URL_2_QUERY_KEY)].filter(Boolean);
const signedUrls = await Promise.all(urls.map(url => {
   const signedUrl = await fetch('API_END_POINT/url')
   return signedUrl;
}))
buildPlayerFromBagURLs(signedUrl)...
```

vidaaudrey, 2020-08-26:
@castaway2000 gonna close the issue now. Let us know if you have any questions

closed 2020-08-26 by vidaaudrey