cruise-automation/webviz #497: npm run bootstrap for static build fails with error

ajayvohra2005, 2020-08-24:
This is for hash commit:  e4af0f2e0a049e4589a6e19fddb71c6f6a86ac8c 
Ubuntu 18.04

```
sudo apt install nodejs npm
```

```
Reading package lists... 0%
Reading package lists... Done
Building dependency tree       
Reading state information... Done
npm is already the newest version (3.5.2-0ubuntu4).
nodejs is already the newest version (8.10.0~dfsg-2ubuntu0.4).
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
```
```
git clone https://github.com/cruise-automation/webviz.git
```

```
Cloning into 'webviz'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 14877 (delta 0), reused 3 (delta 0), pack-reused 14871
Receiving objects: 100% (14877/14877), 159.39 MiB | 28.65 MiB/s, done.
Resolving deltas: 100% (11213/11213), done.
```

```
cd webviz
npm run bootstrap
```

```
npm WARN deprecated url-search-params@1.1.0: now available as @ungap/url-search-params
npm WARN deprecated rollup-plugin-node-resolve@3.4.0: This package has been deprecated and is no longer maintained. Please use @rollup/plugin-node-resolve.
npm WARN deprecated rollup-plugin-replace@2.2.0: This module has moved and is now available at @rollup/plugin-replace. Please update your dependencies. This version is no longer maintained.
npm WARN deprecated rollup-plugin-babel@4.4.0: This package has been deprecated and is no longer maintained. Please use @rollup/plugin-babel.
npm WARN deprecated text-encoding@0.7.0: no longer maintained
npm WARN deprecated rollup-plugin-commonjs@9.3.4: This package has been deprecated and is no longer maintained. Please use @rollup/plugin-commonjs.
npm WARN deprecated chokidar@2.1.8: Chokidar 2 will break on node v14+. Upgrade to chokidar 3 with 15x less dependencies.
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm WARN deprecated core-js@1.2.7: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated @types/vfile-message@2.0.0: This is a stub types definition. vfile-message provides its own type definitions, so you do not need this installed.
npm ERR! Linux 5.3.0-1032-aws
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code EMISSINGARG

npm ERR! typeerror Error: Missing required argument #1
npm ERR! typeerror     at andLogAndFinish (/usr/share/npm/lib/fetch-package-metadata.js:31:3)
npm ERR! typeerror     at fetchPackageMetadata (/usr/share/npm/lib/fetch-package-metadata.js:51:22)
npm ERR! typeerror     at resolveWithNewModule (/usr/share/npm/lib/install/deps.js:456:12)
npm ERR! typeerror     at /usr/share/npm/lib/install/deps.js:457:7
npm ERR! typeerror     at /usr/share/npm/node_modules/iferr/index.js:13:50
npm ERR! typeerror     at /usr/share/npm/lib/fetch-package-metadata.js:37:12
npm ERR! typeerror     at addRequestedAndFinish (/usr/share/npm/lib/fetch-package-metadata.js:82:5)
npm ERR! typeerror     at returnAndAddMetadata (/usr/share/npm/lib/fetch-package-metadata.js:117:7)
npm ERR! typeerror     at pickVersionFromRegistryDocument (/usr/share/npm/lib/fetch-package-metadata.js:134:20)
npm ERR! typeerror     at /usr/share/npm/node_modules/iferr/index.js:13:50
npm ERR! typeerror This is an error with npm itself. Please report this error at:
npm ERR! typeerror     <http://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /home/ubuntu/webviz/npm-debug.log

npm ERR! Linux 5.3.0-1032-aws
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "bootstrap"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! root@ bootstrap: `npm install && lerna bootstrap --hoist "{react,react-dom}"`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the root@ bootstrap script 'npm install && lerna bootstrap --hoist "{react,react-dom}"'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the root package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     npm install && lerna bootstrap --hoist "{react,react-dom}"
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs root
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls root
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/ubuntu/webviz/npm-debug.log
```

vidaaudrey, 2020-08-24:
The node and npm version looks a bit old. Could you try upgrade? My current npm version is 6.13.4 and node version is 10.16.3.

zlg9folira, 2020-08-25:
I had the same issue, upgraded node.js and npm  to 14.4 and 6.14:
`node -v : v14.4.0`
installed at:
`which npm: /home/user/.nvm/versions/node/v14.4.0/bin/npm`
`npm -v : 
{
  npm: '6.14.8',
  ares: '1.16.0',
  brotli: '1.0.7',
  cldr: '37.0',
  icu: '67.1',
  llhttp: '2.0.4',
  modules: '83',
  napi: '6',
  nghttp2: '1.41.0',
  node: '14.4.0',
  openssl: '1.1.1g',
  tz: '2019c',
  unicode: '13.0',
  uv: '1.37.0',
  v8: '8.1.307.31-node.33',
  zlib: '1.2.11'
}
`

Now by running `npm run bootstrap`, build fails with too many errors including `error: no matching function for call to ...`:

`gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/user/webviz/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Linux 4.15.0-112-generic
gyp ERR! command "/home/user/.nvm/versions/node/v14.4.0/bin/node" "/home/user/webviz/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--verbose" "--libsass_ext=" "--libsass_cflags=" "--libsass_ldflags=" "--libsass_library="
gyp ERR! cwd /home/user/webviz/node_modules/node-sass
gyp ERR! node -v v14.4.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
Build failed with error code: 1
npm WARN @dump247/storybook-state@1.5.2 requires a peer of @storybook/addons@^3.2.16 but none is installed. You must install peer dependencies yourself.
npm WARN acorn-jsx@5.0.1 requires a peer of acorn@^6.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN jest-puppeteer@4.3.0 requires a peer of puppeteer@^1.5.0 but none is installed. You must install peer dependencies yourself.
npm WARN redux-devtools-extension@2.13.7 requires a peer of redux@^3.1.0 || ^4.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN storycrawler@2.3.5 requires a peer of puppeteer@^1.19.0 but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/webpack-dev-server/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/watchpack/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/watchpack-chokidar2/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-sass@4.10.0 postinstall: `node scripts/build.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-sass@4.10.0 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/user/.npm/_logs/2020-08-25T17_01_40_073Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! root@ bootstrap: `npm install && lerna bootstrap --hoist "{react,react-dom}"`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the root@ bootstrap script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/user/.npm/_logs/2020-08-25T17_01_40_153Z-debug.log
`

vidaaudrey, 2020-08-25:
We haven't tested with higher node versions yet. @zlg9folira  could you use nvm to try node v.10.16.3 and see if that works for you?

zlg9folira, 2020-08-25:
Node v10.16.3 with 6.9.0 worked. All builds passed:
`npm run bootstrap # install dependencies`
`npm run build # build all packages`
`npm run build-static-webviz # generate static build in __static_webviz__`
`npm run serve-static-webviz # serve static build on localhost:8080`

Thanks

closed 2020-08-26 by vidaaudrey