cruise-automation/webviz #498: Error: Unsupported encoding 8UC3 for sensor_msgs/Image msgs in Image panel

ajayvohra2005, 2020-08-24:
Raw  messages received on ```sensor_msgs/Image``` data type topic are displayed fine in webviz, but when viewing these messages in a webviz Image panel, I see the error **Error: Unsupported encoding 8UC3**.  ```rosbridge``` server is using ros noetic distro running in ubuntu focal.

davidcrawford, 2020-08-24:
Hi Ajay, could you provide a sample bag to reproduce the issue?

ajayvohra2005, 2020-08-25:
I am unable to share a sample bag but I can share in detail what I am doing. 

Steps outline:
* Converting using ```cv2_to_imgmsg``` with pass through encoding
* I am writing to a bag file
* I am publishing the bag file 
* Reading the ros topics in ```webviz``` via ```rosbridge```

```
ros_image_msg = img_cv_bridge.cv2_to_imgmsg(image_data)
ros_image_msg.header.stamp.secs = divmod(image_ts[i], 1000000 )[0] #stamp in micro secs
ros_image_msg.header.stamp.nsecs = divmod(image_ts[i], 1000000 )[1]*1000 # nano secs
bag.write(ros_topic, ros_image_msg, ros_image_msg.header.stamp)
```

vidaaudrey, 2020-08-26:
Hi @ajayvohra2005 we currently support a handful of encodings which doesn't yet include `8UC3`: https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ImageView/renderImage.js#L11-L20

Maybe you could try using other encoding format? Also feel free to open a PR to add additional decoding support :)