cruise-automation/webviz #502: Failed to decode image on ios chrome

XinnWang, 2020-09-01:
Hi, thanks for the great work!

i want to open the Webviz on my iphone or ipad. i am using the chrome but got following warning when opening the url:

[image]

after choosing the "continue anyway", i got following error:

[image]

it said "webviz failed to decode image". What should i do ? Thanks.

troygibb, 2020-09-01:
Thanks @XinnWang! We don't officially support iOS or iPadOS-- so there may be an issue. It would be also helpful to know whether this works on your desktop browser.

XinnWang, 2020-09-01:
@troygibb  is Android supported?

troygibb, 2020-09-01:
Not officially.

closed 2020-09-07 by XinnWang