cruise-automation/webviz #503: visualization_msgs/MarkerArray is not rendered correctly on Android Chrome

muraki-t4, 2020-09-04:
Hi, Team.

Thank you for your great work :)

I'm trying to render the vector map (like the following `visualization_msgs/MarkerArray` messages) in a 3D panel **on Android Chrome**.

But the map is not rendered correctly on only one device even though each Android Chrome version is same.

I’m not sure but I think this issue was caused by the difference of chip set on Android.

**Can you think of anything that might have caused it?**

_Android Phone(Qualcomm® Snapdragon™ 670): **Chrome 85.0.4183.81** <- OK_

[image]

_Android Tablet(MediaTek Helio P22T): **Chrome 85.0.4183.81** <- NG_

[image]

Here is an extract of a part of the `visualization_msgs/MarkerArray` messages. (Sorry, I can't share whole messages.)

```
markers: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "map"
    ns: "road_edge"
    id: 0
    type: 4
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.08
      y: 0.0
      z: 0.0
    color: 
      r: 0.5
      g: 0.5
      b: 0.5
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: True
    points: 
      - 
        x: -47240.59
        y: 57656.569
        z: 24.219
      - 
        x: -47240.21
        y: 57656.878
        z: 24.216
      - 
        x: -47227.883
        y: 57622.15
        z: 24.466
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "map"
    ns: "road_edge"
    id: 1
    type: 4
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.08
      y: 0.0
      z: 0.0
    color: 
      r: 0.5
      g: 0.5
      b: 0.5
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: True
    points: 
      - 
        x: -47314.123
        y: 57614.194
        z: 24.793
      - 
        x: -47314.838
        y: 57613.496
        z: 24.799
      - 
        x: -47324.134
        y: 57624.443
        z: 24.773
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
---
```

vidaaudrey, 2020-09-04:
@muraki-t4 could you perhaps create a codesandbox with the marker array data so we could try to reproduce in different devices? We haven't been supporting mobile devices in the past. Would be interesting to know what are the limits so we could plan for our support accordingly.

muraki-t4, 2020-09-09:
@vidaaudrey Sorry for my late response. Is threre any templates of codesandbox for webviz?

muraki-t4, 2020-09-11:
This issue is caused by GPU chip (IMG PowerVR GE8320).
I confirmed it is not occurred on GPU chip (Qualcom Adreno).

closed 2020-09-11 by muraki-t4