cruise-automation/webviz #520: change localhost to ip

sky4133, 2020-09-29:
How can I change the localhost to a ip address?
I want to browse the webviz program from the ip address , 
and then , I can browse it in another computer .

troygibb, 2020-09-29:
Hello! Check out this [stack overflow answer](https://stackoverflow.com/questions/13322485/how-to-get-the-primary-ip-address-of-the-local-machine-on-linux-and-os-x). That'll give you some tools on how to ascertain your machine's ip in your network.

closed 2020-09-29 by troygibb

sky4133, 2020-09-30:
Thanks for your answer， now I can browse the webviz with the machine's IP like,http://192.168.0.90:8080/app/, and also the website can be browsed  from another computer in the same local area network. 
But when I browse the website from another computer，the webviz can't receive the Radar Topic an data.
Could you give me some advise?

polleyr, 2020-12-09:
Set up your rosbridge to use the IP address of the host PC in your network (so 141.XX.XX.XX:9090 and not 0.0.0.0:9090 for example). You can use [this launchfile](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_server/launch/rosbridge_websocket.launch), just edit the 3rd line.

 Then you can use 141.XX.XX.XX:8080/?rosbridge-websocket-url=ws://141.XX.XX.XX:9090 in order to connect the rosbridge to webviz.