cruise-automation/webviz #522: change color of radar points in 3D panel

sky4133, 2020-09-30:
In my Webviz, the color of online radar points is only white with black background,
how could I change the color to "Rainbow" like the Demo from Webviz.

Ps, when the key "colorMode", which is copied from Webviz demo, was add into Json of panel setting , nothing changed.
  "settingsByKey": {
    "t:/rslidar_points": {
      "colorMode": {
        "mode": "rainbow",
        "colorField": "ring"
      }
    }
}

vidaaudrey, 2020-09-30:
@sky4133  if you color by `ring`, you'll need to make sure this field is actually available in the lidar data. Otherwise, you can always choose to color by other available fields. 
[image]

sky4133, 2020-09-30:
thank you !!! it works

closed 2020-10-19 by vidaaudrey