cruise-automation/webviz #523: Browse the webviz in LAN or WLAN

sky4133, 2020-09-30:
Now I can browse the webviz with the machine's IP like,http://192.168.0.90:8080/app/, and also the website can be browsed from another computer in the same local area network.
But when I browse the website from another computer，the webviz can't receive the Radar Topic an data.
Could you give me some advise?

sky4133, 2020-10-09:
May anybody help me ?

vidaaudrey, 2020-10-19:
Hi @sky4133, are you trying to view the radar topic from websocket or a bag? Is the topic turned on (e.g. in the topic tree of 3D panel or Raw Messages panel) in the other computer that you are trying to browse from?

sky4133, 2020-10-23:
hi vidaaudrey， I can browse the webviz with Lidar data in the local computer, which can connect the lidar . But the Lidar data can only send to the local comuputer,  it can not be transform to another computer in the same LAN . Now only the webviz can be browsed from LAN without Lidar data.

closed 2021-01-19 by sky4133