cruise-automation/webviz #538: Error playback demo.bag in clean chrome

Uezzi, 2020-10-29:
First congratulation for the great Webviz project! However, I have an issue testing the demo.bag file (downloaded [here](https://open-source-webviz-ui.s3.amazonaws.com/demo.bag)). I'm using google chrome 86.0.4240.111 after clearing the complete history including caches.

If I browse the Webviz page (e.g. the [hosed one](https://webviz.io/app)) and drop the demo.bag file there and activate the topic /velodyne_points, the following error occurs: missing transforms to root transform map.
If I browse the demo, using the ?demo prefix, everything looks fine. After browsing the ?demo I'm going back to the [app page](https://webviz.io/app). If I drop the demo.bag file again, everything looks fine, too. The same issue can be seen if using Inkognito mode in chrome.

vidaaudrey, 2020-10-29:
Hi @Uezzi, if you change the follow option to `velodyne` at the right side of the 3D panel, you should see the the markers showing up. Let me know if you still have issues. 

[image]

Uezzi, 2020-11-02:
Hi @vidaaudrey , it works now. Thanks for your help!

closed 2020-11-02 by Uezzi