cruise-automation/webviz #552: Node Playground: simple example for sum two signals

gismo07, 2020-11-19:
We are looking for a small example to sum two signals and publish a simple Float32 message inside the node playground.

troygibb, 2020-11-19:
Here you go! The key is to sync the two topics as `publisher` is invoked:

```
import { Input, Messages } from "ros";
import { areSame } from "./time";

export const inputs = ["/signal/a", "/signal/b"];
export const output = "/webviz_node/sum";

type Message = Input<"/signal/a"> | Input<"/signal/b">;

const msgsByTopic: { [topic: string]: Message } = {};

const publisher = (message: Message): { sum: number } | undefined => {
  msgsByTopic[message.topic] = message;
  const signalOne = msgsByTopic["/signal/a"];
  const signalTwo = msgsByTopic["/signal/b"];

  // Don't return if the two topics aren't synced.
  if (!signalOne || !signalTwo || !areSame(signalOne.receiveTime, signalTwo.receiveTime)) {
    return;
  }

  return {
    sum: signalOne.message.value + signalTwo.message.value
  };
};

export default publisher;
```

gismo07, 2020-11-19:
Thank you for the quick response. Ill check it out