cruise-automation/webviz #557: Webviz fails to initialize when plots are being displayed

zhihao95, 2020-12-08:
I'm currently having a issue when intializing webviz with the layout containing plots. This seems to only happen to Windows as i tried it on Ubuntu and it works fine. 

[image]

zhihao95, 2021-02-01:
Are there any updates regarding this?

hassoncs, 2021-02-10:
Hi @zhihao95, I'm not able to reproduce the problem. Could you post the version of chrome you are using? Also, are there any JS errors in the console?

Here's a generic page on the error, but I don't think it's much help: https://support.google.com/chrome/answer/95669

zhihao95, 2021-02-11:
I'm using Chrome version 88.0.4324.150 (Official Build) (64-bit). Also, we are using remote bag and remote layout to load the bags and layouts. 

I couldn't get the JS error to show using chrome (it just crashes the page and shows either STATUS_ACCESS_VIOLATION or STATUS_NOT_IMPLEMENTED)

However, this shows if I use firefox instead 

[image]