cruise-automation/webviz #568: Support mesh MarkerArray type

chuan9, 2020-12-23:
Hi, all
I noticed that the webviz doesn't support mesh markerarray type. Do you have any plans to support it?

ThreeDimensionalViz/SceneBuilder/index.js
```
      case 9: return add.text(marker);
      // mesh resource not supported
      case 11: return add.triangleList(marker)
```

chuan9, 2020-12-23:
@vidaaudrey

vidaaudrey, 2021-01-05:
@chuan9 we don't have any plans to support it since we are not using it internally. I'm happy to work with you to add support if you are interested 🙂

droter, 2021-01-09:
@vidaaudrey I am interested in helping with the mesh feature.

vidaaudrey, 2021-01-11:
@droter that's great! Below are some links that might be helpful:
- build the command to render the mesh marker. You could refer the commands [here](https://github.com/cruise-automation/webviz/tree/master/packages/webviz-core/src/panels/ThreeDimensionalViz/commands). 
-  [add](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/hooksImporter.js#L100-L112) the datatype to the `SUPPORTED_MARKER_DATATYPES` list to allow the msgs to be passed down to the SceneBuilder
- add the logic to [consume](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/SceneBuilder/index.js#L775) the mesh messages in SceneBuilder
- collect the new mesh markers by [adding](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/SceneBuilder/index.js#L946) case 10 in `_addMarkerToCollector`
- [render](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/WorldMarkers.js#L215-L284) the markers in `WorldMarkers`

Let me know if you come across any issues :)

chuan9, 2021-01-12:
@vidaaudrey Sorry, I'm intrested in this feature but have no time. 
@droter Waiting for your contribution

luqnaim, 2021-06-16:
Upvote on this. Encountered the same unsupported marker type error. Would be nice to render mesh objects in the 3D panel!