cruise-automation/webviz #570: Topic "/topic_name_here" has bad frame / wrong visualization of visualization_msgs/MarkerArray

marcelino-pensa, 2020-12-29:
Hello, world! First of all, I'd like to thank you for this awesome tool, it is really helpful to visualize our data without having to rely on Rviz, so thanks for that!

I've been trying to figure out how to use webviz, and things were going on smoothly other than I'm getting the following error for certain topics:
[image]

It seems that some topics that suffer from this problem (I have only seen it in topics of the type `visualization_msgs/MarkerArray`) end up being published somewhat based on the root of the tf tree. Since these topics are not being published in the root of the `tf` tree, then weird stuff starts happening.

As an example, here is how I visualize a `CUBE_LIST` in `webviz` (left) vs how it is visualized in `Rviz (right)`:

[image]

It turns out that the root of my tf tree is rotated by 180deg with respect to the frame_id that I publish my MarkerArray, so it seems like webviz is using the root frame to visualize the topic, while it is being published in a different frame_id.

As an extra piece of information, not all `visualization_msgs/MarkerArray` topics are mirrored like this. I have noticed that it is weird if the topic is of the type:
- CUBE_LIST
- SPHERE_LIST

If the topic is of the following types, then **it has no problems** (it uses the correct frame to visualize):
- LINE_LIST
- CUBE

I'm not sure if I'm doing something wrong on my side, or if the issue is with webviz. All I know is that I can visualize correctly with rviz, but I can't do so with webviz. I'd appreciate any feedback on that front! =)

vidaaudrey, 2021-01-05:
@marcelino-pensa thanks for reporting. As far as I know, we don't do anything special between cube/line and cube_list/line_list before rendering it in [WorldMarkers](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/WorldMarkers.js#L229-L245).  Could you record a bags with the working and non-working marker types so we could help further debug?

marcelino-pensa, 2021-01-05:
Hi @vidaaudrey , thank you for your response!
You can download a bag from the following link: https://drive.google.com/drive/folders/1F8XBAdPvk9xGe90zGTYZJu4fA6aql5Hn?usp=sharing

The link above also has an `.rviz` config for you to visualize the data within `Rviz`, as well as a `.json` config for webviz. 
In summary, the bag has the following topics:
- `/tf`
- `tf_static`
- `/mapper_node/inflated_obstacle_markers` - CUBE_LIST
- `/mapper_node/path_planning_config_markers` - CUBE

You will notice that once you play the bag in `webviz` you'll have the topic `/mapper_node/inflated_obstacle_markers` mirrored with respect to Rviz, while `/mapper_node/path_planning_config_markers` isn't mirrored.

vidaaudrey, 2021-01-05:
Thanks for providing more details. This is a known issue where marker messages are rendered using the most recent tf frame instead of the header.stamp's frame id. This requires significant change on our end and we de-prioritized it. The workaround is to[ render all makers using the rootTransformId](https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/ThreeDimensionalViz/SceneBuilder/index.js#L444-L452).

marcelino-pensa, 2021-01-06:
Thanks for the response! I have been using the workaround, it is not ideal but it has been working fine!

Is there another open issue for the problem above? I could close the current issue mentioning the described issue to prevent duplicates

vidaaudrey, 2021-01-06:
Thank you. We have an internal ticket tracking it. Let's keep it open for now and we could try to fix if we have enough resources and use asks.

marcelino-pensa, 2021-01-06:
Thanks! Now that I think of it, is there any good explanation for why `CUBE` renders correctly but `CUBE_LIST` doesn't? Based on the description above, either both should work or either should fail. Am I misinterpreting this somehow?

Mast3rChief, 2021-02-26:
Hi, I'm also experiencing the same problem. I'm trying to visualize a marker_array with map information for my project. The transformation error makes the visualization result quite ugly. Lines are very distorted.

I'd very much appreciate this fix.

liespace, 2021-12-30:
same problem, hope the fix