cruise-automation/webviz #573: Worldview GLTFScene errored when rendering, (regl) unknown parameter (undefined).

nutchun, 2021-01-01:
I'm using Worldview in my React project. When I want to render GLTFScene in Worldview component, I got an error like this:
```
Error: (regl) unknown parameter (undefined). possible values: mipmap,nearest mipmap nearest,linear mipmap nearest,nearest mipmap linear,linear mipmap linear,nearest,linear
```
I have used DamagedHelmet model ([https://github.com/KhronosGroup/glTF-Sample-Models/tree/master/2.0/DamagedHelmet](https://github.com/KhronosGroup/glTF-Sample-Models/tree/master/2.0/DamagedHelmet)) to test this out but I got an error when rendering. It seems like the argument `value` in glConstantToRegl from GLTFScene commands received `undefined` value.
Any help for me, thanks in advance.

vidaaudrey, 2021-01-05:
Looks like we are getting `undefined` texture here since we can't get the correct sampler data from `model.json.sampler`. I uploaded the model to https://gltf-viewer.donmccurdy.com/ and was able to see it. @jtbandes created `GLTFScene` initially, maybe you know what's the best way to add support for this model format?

[image]

hassoncs, 2021-01-05:
@vidaaudrey I think we might need to update `GLTFScene.js` to honor the defaults for the each sampler property. According to this schema I found, it seems like it's valid to have an "empty" sampler which should fall back to the defaults for `magFilter `,`minFilter`,`wrapS`, and `wrapT`.

https://github.com/KhronosGroup/glTF/blob/master/specification/1.0/schema/sampler.schema.json

hassoncs, 2021-01-05:
Yeah, I think the fix is we should not set those properties (min/mag/wrapS/wrapT) on the `regl.texture` if they aren't present in the `sampler`.

vidaaudrey, 2021-01-05:
I did try to filter out the undefined texture but end up not seeing the model in the scene. Didn't go deep though. This is different from falling back to default right?

hassoncs, 2021-01-07:
@vidaaudrey, this was the fix I was thinking of: https://github.com/cruise-automation/webviz/pull/576 I haven't gotten a chance to test it out yet though

vidaaudrey, 2021-01-08:
@hassoncs  probably still missing something: 
[image]

The https://gltf-viewer.donmccurdy.com/ is showing the model correctly:
[image]