cruise-automation/webviz #588: Feature Request: Allow publish pane to publish std_msgs::Empty

marcelino-pensa, 2021-02-04:
I feel like one of the most useful features of buttons in GUIs is to just send a trigger request. This can be easily made through a `std_msgs::Empty` publish in a given topic, but this isn't available in webviz.
It isn't a big deal not to be able to publish `std_msgs::Empty` because we can publish `std_msgs::Bool` instead. However, I'll leave this issue here in case y'all want to tackle it sometime in the future

jhurliman, 2021-05-20:
@marcelino-pensa the only natively supported datatypes in webviz are 3D markers and 3D marker arrays, I believe. You are likely seeing some of the datatypes in that list because other running nodes are publishing topic using those types. I just merged a change into Foxglove Studio to support the majority of `ros/common_msgs` datatypes out of the box.

marcelino-pensa, 2021-05-20:
Thanks, @jhurliman!
I just ran a dummy node that had one publisher and one subscriber to `std_msgs/Empty`, but webviz did not list it as an option for a data type... it might be that some part of `webviz's` pipeline does not like an empty message definition, and it might be a bit confused with that...
But you are right, I have made some custom msg definitions (non-standard ros messages), and webviz shows them with no problem, the issue just seems to be with `Empty` itself
As for your merge, when should we expect it to get into `https://webviz.io/app/`? Or is it supposed to show already?

marcelino-pensa, 2021-05-20:
Ooh, I have just realized that Foxglove is a fork of `webviz`! I'll take a look at it (and perhaps move our applications to Foxglove) when I get some time
Thanks for the feedback, though!