cruise-automation/webviz #589: How can I use Webviz to display a Path type Message？

Chenms0110, 2021-02-18:
For Path type Message, there is no way for me to successfully draw a picture in the 3d display interface, and it is recognized as MarkerArrays. 
How can I display Path type Message?

cjds, 2021-03-01:
Seconded `[nav_msgs/Path](http://docs.ros.org/en/api/nav_msgs/html/msg/Path.html)`  would be great to have in [the](https://github.com/cruise-automation/webviz/blob/0bef2b84f3e20831570202f5baf3f173785e026d/packages/webviz-core/src/util/globalConstants.js#L43) list

cjds, 2021-03-04:
@Chenms0110 I have a patch here that adds support for it here https://github.com/cruise-automation/webviz/pull/592

Chenms0110, 2021-03-05:
@cjds Thanks! I will try it.
Now I have changed my nav_msgs/Path messages to a marker(line strip type) messages with its x, y, z...... 
Webviz can show this Marker message in 3d display interface.

closed 2021-03-05 by Chenms0110