cruise-automation/webviz #598: Unable to get data from multiple topics easily

rickyleal2, 2021-03-15:
I am trying to display data from multiple topics in plot panels and then download all the data in CSV files. I have over 30 topics

I tried using a Tab Panel containing multiple Plot panel per topic but switching tabs erase previous data on the panels.
I also tried using independent Plot panels but having too many plot panels does not allow me to visualize the data properly.
I also tried having a single plot panel with all the topics but this is not ideal. Moreover, there is a point where I can't add more topics to the plot as the list of topics gets too long to fit on the screen.

Is there any way I can keep the data on the plots after switching tabs?

vidaaudrey, 2021-03-15:
You could try using Node Playground to take multiple topics and merge into one before publishing: https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/src/panels/NodePlayground/index.help.md#using-multiple-input-topics

rickyleal2, 2021-03-16:
We tried using Node Playground but that approach did not solve my problem. What would solve my problem is data persistence when swapping between tabs