cruise-automation/webviz #604: Talking to webviz directly from non-ROS application

AndreiBarsan, 2021-03-19:
Hi WebViz team, 

I am interested in using webviz for ad-hoc visualization tasks, e.g., quickly showing point clouds or images from a Python project that does not use ROS.

It seems like it would be possible to communicate with webviz by directly synthesizing, e.g., visualization messages in JSON and writing them to an appropriate websocket, without relying on ROS + rosbridge.

Are there any examples of code doing this? Is this even possible, or is it absolutely necessary to have a full instance of ROS running under the hood for visualizing things in webviz?

Thank you!

kwetril, 2021-03-20:
Have the same problem, trying to implement my own writer for rosbag files without ros dependency for that.

shreyas2311, 2021-04-14:
@kwetril @AndreiBarsan Were you able to achieve this? It would be great to know.

amacneil, 2021-04-14:
What format(s) are your data currently in?

shreyas2311, 2021-04-27:
@amacneil Thanks for the reply. I have my data stored in protobuf format.

AndreiBarsan, 2021-04-27:
@shreyas2311 I was able to get this working by writing ROS messages to a websocket with [roslibpy](https://roslibpy.readthedocs.io/), and routing those to a running ROS instance via `rosbridge`, which was the instance that `webviz` connected to.

So this was still in a way relying on ROS, so not super clean. I was running ROS in its own container but this still had a fair bit of overhead.

amacneil, 2021-04-29:
This is something we've been thinking about for [foxglove/studio](https://github.com/foxglove/studio) (fork of webviz). We really want to add support for non ROS formats.

However, it would really help to understand more specifically how your data is structured. For example, [sensor_msgs/PointCloud2](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/PointCloud2.html) is a standard message for storing point clouds in ROS. If you're storing point clouds in protobuf, what do the definitions look like? How would you expect to tell webviz how to decode your point cloud?