cruise-automation/webviz #623: Error when openning localhost:8080 in my chromium browser after run serve-static-webviz

Nick-PAN, 2021-04-22:
I build manually from the source. 
```
Error stack:
Error: Minified React error #321; visit https://reactjs.org/docs/error-decoder.html?invariant=321 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at Z (http://localhost:8080/webvizCoreBundle.js:359:9310)
    at exports.useEffect (http://localhost:8080/webvizCoreBundle.js:359:11361)
    at exports.useCleanup (http://localhost:8080/webvizCoreBundle.js:95:314652)
    at exports.useMessageReducer (http://localhost:8080/webvizCoreBundle.js:95:310519)
    at Object.exports.useMessagesByTopic (http://localhost:8080/webvizCoreBundle.js:359:14020)
    at http://localhost:8080/hooks_bundle.js:126:61047
    at wh (http://localhost:8080/webvizCoreBundle.js:224:78149)
    at di (http://localhost:8080/webvizCoreBundle.js:224:85444)
    at ci (http://localhost:8080/webvizCoreBundle.js:224:85265)
    at ai (http://localhost:8080/webvizCoreBundle.js:224:84984)
Component stack:
    in Unknown
    in RosoutPanel
    in ErrorBoundary
    in div
    in Flex
    in Panel(RosoutPanel)
    in div
    in div
    in InternalMosaicWindow
    in DragSource(InternalMosaicWindow)
    in DropTarget(DragSource(InternalMosaicWindow))
    in MosaicWindow
    in div
    in div
    in MosaicRoot
    in div
    in MosaicRoot
    in ErrorBoundary
    in UnconnectedPanelLayout
    in ForwardRef
    in div
    in Provider
    in Unknown
    in Unknown
    in ConnectFunction
    in div
    in Unknown
    in Unknown
    in ComponentWithDragDropContext
    in ConnectFunction
    in Route
    in ErrorBoundary
    in div
    in Router
    in ConnectedRouter
    in ConnectedRouterWithContext
    in ConnectFunction
    in Provider
    in Unknown
    in Root
```

Nick-PAN, 2021-04-22:
Using Chrome browser, it worked!

closed 2021-04-22 by Nick-PAN

ovoayu, 2024-07-16:
Hello, I am using chrome browser, but I still have the same problem, do you have any other solution