cruise-automation/webviz #650: failed to run in docker or remote, report Soleil.woff2 not found

HugoFeng, 2021-06-24:
`172.17.0.1 - - [24/Jun/2021:09:09:52 +0000] "GET /res/Soleil.woff2 HTTP/1.1" 404 555 "http://localhost:8080/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.77 Safari/537.36" "-"

2021/06/24 09:14:21 [error] 32#32: *4 open() "/usr/share/nginx/html/res/Soleil.woff2" failed (2: No such file or directory), client: 172.17.0.1, server: localhost, request: "GET /res/Soleil.woff2 HTTP/1.1", host: "localhost:8080", referrer: "http://localhost:8080/"`

I ran with "docker run -p 8080:8080 cruise/webviz", and got this in the command line, the localhost:8080 only appear a logo flashing forever. 
I try to inspect, chrome console shows: 
`Failed to load resource: the server responded with a status of 404 (Not Found)

loadWebviz.js:241 Uncaught (in promise) DOMException: A network error occurred.`

When I open "https://webviz.io/app/", I got the same problem and console feedback.

HugoFeng, 2021-06-25:
Problem solved, seem like it's related to some extensions in my Chrome. I start with a clean chrome and everything runs fine.

closed 2021-06-25 by HugoFeng

RiZKiT, 2021-12-02:
In the case someone stumbles upon this, is seems to be https://chrome.google.com/webstore/detail/evernote-web-clipper/pioclpoplcdbaefihamjohnefbikjilc